lumapu/ahoy #1658: MQTT JSON-Payload active_PowerLimit wird nicht mehr ausgegeben

knickohr, 2024-06-02:
Bein neuen JSON-Payload ab der 0.8.123 wird das active_PowerLimit der Inverter nicht mehr mit ausgegeben.

<topic>/<inverter-name>/ch0/active_PowerLimit fehlt in dem JSON-String

geronet1, 2024-06-07:
Ist jetzt eine Eben höher, nicht mehr in /ch0/

knickohr, 2024-06-07:
Das dachte ich mir erst auch, habe es aber nicht gefunden. Macht da ja auch Sinn. Jetzt wo Du es bestätigst werde ich einfach mal mithören 😉

knickohr, 2024-06-07:
Nein, kommt bei mir leider nicht. Du meinst nicht dieses ack_power_limit ?  Ich meine active_PowerLimit.

knickohr, 2024-06-07:
Axo, jetzt mit dem neuen PR 😅

knickohr, 2024-06-10:
Leider offenbar nicht aktiv, ich kann den Topic nicht finden 😢

geronet1, 2024-06-11:
Das ist nicht retained, heißt es kommt einmal beim Start und wenn man einen neuen Wert setzt.

closed 2024-06-12 by knickohr