lumapu/ahoy #1664: Feature Request: Calculate daily History plot from Sunrise/sunset

BenjiKimba, 2024-06-07:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

BenjiKimba, 2024-06-07:
Calculate daily History plot from Sunrise/Sunset

BenjiKimba, 2024-06-07:
[image]

And also to improve the resolution of the history curve.

BenjiKimba, 2024-06-07:
[image]
This is how it looks at 16:36

Gubi2023, 2024-06-07:
hello, the latest dev.  Versions have time Stamps in history, see screenshot   
[image]

Gubi2023, 2024-06-07:
otherwise you have to wait for the next release...

BenjiKimba, 2024-06-07:
Ah, thank you.

closed 2024-08-16 by lumapu