lumapu/ahoy #1665: Fehlende Menü Eintäge /  Darstellung in der Web GUI, Warum?

ralfberger, 2024-06-07:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ x] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.123

Ich bekomme nur diese Darstellung:
[image]

das restliche Menü fehlen, egal ob ich  FireFox oder Chrome nutze.

Gubi2023, 2024-06-07:
Die Verbindung zur DTU ist unterbrochen. Entweder warten bis sie aufgebaut wird, oder DTU aus/einstecken

ralfberger, 2024-06-07:
Ich verstehe deine Antwort nicht, wie meist du das? Ich habe doch eine Verbindung zum Webinterface, mein MQTT Broker empfängt auch fleißig alle Werte. Neustart bringt keine Veränderung. Das heißt für mich, die Verbindung zum Wechselrichter ist i.O. und die Verbindung von meinen PC zum ESP funktioniert auch.

dknubben, 2024-06-07:
So sieht die Darstellung aus, wenn Javascript deaktiviert ist oder aus anderen Gründen blockiert wird.

lille2000, 2024-06-07:
Selbe Thema wie hier #1619

ralfberger, 2024-06-07:
@dknubben 
Danke für den Hinweis, aber an Javascript liegt es nicht. Ich habe sowohl verschiedene Browser und auch Rechner ausprobiert. 
Jetzt habe ich den ESP mehrmals Stromlos gemacht und jetzt bekomme ich alle Daten wieder angezeigt.

juepi, 2024-06-08:
Ich kenne das Problem von meinem Setup (auch ESP8266), ich hab schlechten WLAN Empfang und bin mir ziemlich sicher dass es daran liegt.

ralfberger, 2024-06-09:
Das kann sicher auch am schlechten Wlan Empfang liegen.  Bei mir ist es aber bestimmt so wie es lille2000 verlinkt hat.
[#1619](https://github.com/lumapu/ahoy/issues/1619)
Nach dem downgrade auf 0.8.97 läuft die DTU wieder stabil.

closed 2024-06-09 by ralfberger

lumapu, 2024-06-09:
wenn es ein ESP32 ist und dieser die Webserver-Ausfallerscheinungen gibt, wäre es sehr hilfreich, wenn ich dazu einen Coredump bekommen könnte. Unter `/system` gibt es dafür einen Button.

ralfberger, 2024-06-09:
nein, ich nutze ein 8266.