lumapu/ahoy #1668: Tooltips für alle Einstellungen

DanielR92, 2024-06-07:


lumapu, 2024-06-09:
Schwierige Frage. Erst habe ich nicht verstanden was du meinst, aber jetzt glaube ich kann ich dir folgen. Du will im Tooltip einen Link haben, der direct auf https://docs.ahoydtu.de verlinkt?

Ich würde direkt verlinken, und auf der docs Seite schon ein "Template" anlegen, dass dann noch Inhalt bekommen kann.

DanielR92, 2024-06-09:
Aber du hast es verstanden. Richtig, ich wollte gerne schonmal alles vorab alles verlinken und auf eine "blanke" Seite umleiten lassen (mit dem Hinweis das es noch gefüllt wird). 

Das kann ich machen wenn ich Leerlauf habe. ;)

stefan123t, 2024-11-03:
@DanielR92 Du kannst die Seite ja mit den leeren Überschriften als Markdown anlegen und dann ein Inhaltsverzeichnis erstellen. Ich meine dass dann auch Links bzw. Reference Anchors zu den Überschriften angelegt werden, die Du in / anstelle der Tooltips nutzen kannst. Was dann unter dem jeweiligen Punkt steht, bzw. wo dieser Punkt in der Dokumentation dann auftaucht ist ja dann (wenn alles auf einer Seite ist!) noch beliebig anzupassen.