lumapu/ahoy #1672: ESP32 board suggestion

juepi, 2024-06-10:
This is more a basic question: I'm planning to migrate my AhoyDTU from ESP8266 (Lolin D1 Mini) to a ESP32 based board.

I have a bunch of ESP32-S2 (Lolin S2 Mini) boards here, however this would basically only beef up the SRAM (320kB compared to 160kB on ESP8266), imho the CPU speed is not much higher (240MHz compared to 160MHz).
ESP32-S3 would give 512kB SRAM and a dual core CPU at the same speed as the S2.

So is there a general suggestion on which ESP32 chip to use for AhoyDTU? I'm planning to run max. 2 inverters with it.

Thanks in advance,

Juergen

knickohr, 2024-06-10:
Dual Core is fine, but for further Projekts and functions, use more Flash. S3 with 8MB is good.

juepi, 2024-06-13:
For the sake of completeness: after @knickohr's [comment in a different issue](https://github.com/lumapu/ahoy/issues/1619#issuecomment-2164569012) that one of the upcoming functions which will require 8MB flash is zero-export regulation, i've decided to go with a **ESP32-S2 N16R8** based board (16MB flash / 8MB PSRAM).

closed 2024-06-13 by juepi

ricoshe81x, 2025-05-01:
How is it possible to migrate? I mean, just export the config?