lumapu/ahoy #1675: [Bug] Discovery Config (homeassistant) sends corrupt data

mirceapaciu, 2024-06-12:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT broker was added

### Error description

This kind of data is sent to MQTT for homeassistant discovery:
{"name":"CH1_P_DC\u0000\u0000\u0000\u0000�%@<P�?\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000�%@","stat_t":"solar/ahoydtu/HM600/ch1/P_DC","unit_of_meas":"W","uniq_id":"114182952061_ch1_P_DC","dev":{"name":"HM600","ids":"114182952061","mdl":"HM600","cu":"http://192.168.178.63","mf":"Hoymiles"},"exp_aft":95,"dev_cla":"power","stat_cla":"measurement"}

When falling back to an older software version, the problem disappears.

lumapu, 2024-06-12:
This is already solved in development version - I need to release really soon.
Meanwhile you can download development versions from https://fw.ahoydtu.de/fw/
At least version 0.8.84 will solve this.

veilofsecurity, 2024-07-07:
[GIT SHA: 8cdf364 :: 0.8.127](https://github.com/lumapu/ahoy/commits/8cdf364) still has this issue for MQTT topic AHOT-DTU_TOTAL / config. Example for Total YieldDay:
`{"name":"Total YieldDay\u0000l\u0000�?�&�p �?\u0000\u0000\u0000","stat_t":"inverter/total/YieldDay","unit_of_meas":"Wh","uniq_id":"AHOY-DTU_TOTAL_total_YieldDay","dev":{"name":"AHOY-DTU_TOTAL","ids":"AHOY-DTU_TOTAL","mdl":"AHOY-DTU_TOTAL","cu":"http://<ip>","mf":"Hoymiles"},"dev_cla":"energy","stat_cla":"total_increasing"}`

lumapu, 2024-07-07:
are you really sure that you first cleared everything on homeassistant side? I don't know if the data is rewritten if they already exisit before clicking the send button.

closed 2024-08-16 by lumapu