lumapu/ahoy #1676: 0.8.125 verliert WLAN

dtuuser, 2024-06-12:
Seit 0.8.124 verliert meine DTU mindestens zweimal am Tag das WLAN und ist dann nur noch über den AP erreichbar. Nach Strom weg und wieder an, ist er sofort wieder im WLAN.

rmayergfx, 2024-06-14:
0.8.126 schon mal probiert?
Hatte weder mit 0.8.124 noch x.125 Probleme.  Verkabelung vom Modul mal geprüft?

dtuuser, 2024-06-14:
Ja, 126 hatte die gleichen Probleme.
Aktuell habe ich die 123 drauf und seitdem ist alles stabil.

rmayergfx, 2024-06-14:
Welches Modul, welche Hardware? Ohne technische Daten kann man hier nicht wirklich helfen.
Bitte gewöhnt Euch doch mal an, wenigstens die Eckdaten zum Modul, ESP8266 oder ESP32 und Version noch mit anzugeben!

dtuuser, 2024-06-14:
Sorry. Vergessen.
ESP32-D0WD-V3 mit NRF24 mit externer Antenne,  Kondensator und Display SSD1306.
Läuft seit ca.1 Jahr festverlötet in einem Gehäuse. Nicht geändert. Updates immer über Update in Ahoy. Zweimal musste ich über die Ahoy Seite neu Aufspielen, da in zwei Versionen das Updaten defekt war.

dtuuser, 2024-06-17:
mit 123 alles stabil. daher closed

closed 2024-06-17 by dtuuser