lumapu/ahoy #1677: [Bug] Unbekanntes GRID Profile

georg-x, 2024-06-14:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

NA

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

NA

### Debug Serial Log output

```shell
NA
```

### Error description

Unbekanntes Profil
Bitte erstelle einen neuen Issue auf https://github.com/lumapu/ahoy und kopiere die Rohdaten hinein.

0C 03 20 00 00 0B 08 FC 07 30 00 0F 09 F9 00 01 02 3F 00 32 0A 55 00 08 09 F9 10 00 13 88 12 8E 00 01 14 1E 00 01 20 00 00 01 30 03 02 58 09 CB 07 A3 13 92 12 8E 40 00 07 D0 00 10 50 00 00 01 13 9C 01 90 00 32 60 04 00 01 09 E2 0A 10 08 7E 80 00 00 00 08 44 01 2C 08 A0 09 6F 09 B4 01 2C 90 00 00 01 00 64 B0 00 00 00 01 F4 00 5F 70 02 00 01 27 10 A0 02 00 00 00 00 

Profil ist "AT_TOR_Erzeuger_cosphi=1"

chr-braun, 2024-06-28:
Profil ist DE_VDE4105_2018

georg-x, 2024-06-28:
OK, sind in DE_VDE4105_2018 und AT_TOR_Erzeuger_cosphi=1 vielleicht die gleichen Werte enthalten?
Ich finde dazu z.B.:
https://github-wiki-see.page/m/tbnobody/OpenDTU/wiki/Grid-Profile-Parser

chr-braun, 2024-06-30:
sorry falscher comment meinerseits

closed 2024-08-16 by lumapu