lumapu/ahoy #1679: AhoyDtu not working after adding second inv. and typing wrong serial number

BenjiKimba, 2024-06-16:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0_8.83
_.__
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

After adding a second inverter but typing the wrong serial number, AhoyDtu is not working anymore.
Using the CMT2300A and having HMS 1600-4t everythings working fine, but after adding a second HMS 1600-4t and typing
in a wrong serial number, AhoyDtu is behaving bad.
Everything inside is cleared, after typing in always get an error, AhoyDtu is sometimes Host and sometimes in my network.

[image]

[image]

Looking at the display it looks it always reset itself.

Any idea how to solve this ?

lumapu, 2024-06-21:
During testing I had a similar issue. This was resolved in development version only. Can you please try to install the dev version?
https://fw.ahoydtu.de/fw/dev/

BenjiKimba, 2024-06-23:
I don't now how to flash a dev. version.
But after flashing 0.8.8.3 again with the web installer and typing in the right serial its working fine with 6 x HMS 1600-4t.

Gubi2023, 2024-06-23:
try webflasher from https://fw.ahoydtu.de/tools/webflasher/  here you can choose easily each dev. or release- Version

closed 2024-08-16 by lumapu