lumapu/ahoy #1680: Feature Request: Reboot zu einer selbst auszuwählenden Uhrzeit

Ollipop030, 2024-06-16:
Wäre es möglich, statt des Hakens "Reboot at midnight" ein Feld einzublenden, in der man die Zeit selbst eintragen kann? Hat den Hintergrund, ich möchte gern die DTU neu starten lassen, um einem möglichen Absturz zuvor zukommen, bei mir läuft eine Nulleinspeisung. Nachts ist bei mir aber die WLan Nachtschaltung aktiv, und um Mitternacht versucht die DTU sich dann stundenlang vergeblich zu verbinden. Ich würde sie dann täglich gegen mittag neu starten wollen.
Ich weiß, kann man alles über ioBroker HAs und Konsorten, hab ich aber alles nicht.
Über ein Script am Pi würde das wohl auch gehen, ich hätte die Funktion aber gern an einem Ort in den Settings der DTU. Nächstes Jahr hab ich nämlich bestimmt vergessen, warum die DTU sich mittags neustartet, wenn ich das damit machen würde. :-)

DanielR92, 2024-06-16:
An sich lässt sich das schon machen. 
Hier könnte man das ganze minimal erweitern und es standardmäßig auf 0:00 belassen.  

z.B: <input type="time" id="appt" name="schedReboot" min="00:00" max="23:59" value="00:00" required />
Das müsste aber nicht im tickerMidtnight laufen, sondern besser (wenn es sowas gibt) tickMinute oder so?

lumapu, 2024-06-21:
gehört zu #1199 (unsere Wishlist)

closed 2024-07-19 by Ollipop030