lumapu/ahoy #1682: Version 0.8.126 Restart Inverter keine Funktion aus der Weboberfläche

SteiniZuHaus, 2024-06-19:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.126

### Github Hash

d4d97dc

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- Nur ein Wechselrichter HMS-2000
- openDTUFusion Baord
- 

### Error description

Leider funktioniert in der Version der Restart des Inverters weder aus dem Web noch über MQQT.
Das EIN/AUS-Schalten ist in Ordnung

Gubi2023, 2024-06-19:
duplikat zu #1661  ?

SteiniZuHaus, 2024-06-19:
Hat sich glaube ich erledigt.
Man kann den Restart nur erkennen wenn man die DTU rebootet, nur dann werden die Tageswerte auf den Wert des WR gesetzt.

lumapu, 2024-06-21:
Vor einiger Zeit haben wir eine Funktion in Ahoy eingebaut, die genau dieses Phänomen hervorruft. Ahoy merkt sich über den Tag den Tagesertrag. Das war / ist nötig, vor allem wenn am Abend Wolken aufziehen und der Wechselrichter ab und zu abschaltet (zu wenig DC-Leistung).

DanielR92, 2024-06-29:
also issue closed?

closed 2024-08-16 by lumapu