lumapu/ahoy #1683: Duration of sending and receiving a packet / Duty cycle

michaelotto, 2024-06-19:
## Hardware
  * [X] ESP32

### Antenna:
* [X] external antenna (SMA)

I have a question regarding the general protocol: How long does sending a request to the inverter take (in milliseconds)? For how long does the inverter send back its information (in milliseconds)?

I'm a asking because strictly, it isn't legally allowed to send more than 1% of an hour in the 868 MHz band (called "duty cycle").

So for example, if I set the interval of how often the inverter should be updated to 5 seconds, then the combined sending and receiving of data must not take longer than 50 ms. (3600 seconds * 1% = 36 seconds of "sending time" per hour / 720 updates per hour = 50 ms).

Loetnase, 2024-06-19:
Maybe that gives you a first idea, first set this 2 hocks in the System for the logging
[image]
and then lock for the logging in the Web Konsole. You see there the transmit time, the recive time and the kompleet loop time.
Then take your intervalltime and ........ 
[image]
I hope that will help you.

michaelotto, 2024-06-19:
Most of this time it looks like it's after TX and waiting for/processing RX. Not sure one can see how much time was really spent doing "on air time".

If it really were around 210 ms, then the maximum allowed polling interval would be around 20-25 seconds.

knickohr, 2024-06-19:
These are Idle times how Long the DTU is waiting for an RX or for the conplete loop. The real TX is a way shorter.

stefan123t, 2024-11-03:
@michaelotto please take a look at the official NRF24L01+ datasheet from Nordic Semiconductors:

**nRF24L01+ Single Chip 2.4 GHz Transceiver Preliminary Product Specification**
https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf

[image]

Note that we are using the chip in its 250kpbs mode.

For the CMT2300A it is probably similar but on the 868 MHz band instead, so probably 2-3 times.