lumapu/ahoy #1684: PowerLimit nimmt nur ganzzahlige Prozentzahlen an

VWthefirst, 2024-06-21:
Moin moin,

Ich baue seit einiger Zeit an einer dynamischen Batterie Ein- und Aus-Speisung. Nun habe ich heute auf die neuste Firmware geupdated und meine in Watt gesendeten API Limit Befehle werden auf die nächste Prozent Ganzzahl abgerundet. Also ich schicke z.b. 20Watt und gesetzt werden 5% 18Watt. Auch per Hand erhalte ich das gleiche Ergebnis. Außerdem kann ich Werte über 100% schicken und offenbar werden sie auch angenommen.  Könnt ihr mir weiterhelfen?

Vielen Dank im Voraus!
Lieben Gruß
Vin

## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______HM-350
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [x] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

lumapu, 2024-06-21:
welche Version verwendest du? In der aktuellen Dev-Version geht ganz sicher eine Stellen nach dem Komma

VWthefirst, 2024-06-22:
ich habe gerade noch einmal die DTU neu aufgesetzt mit dem Online installer (0.8.83). Weiterhin mit dem selber Ergebnis, siehe Anhang. 

[image]

knickohr, 2024-06-23:
In den neueren Versionen ist es durchaus möglich 😉

[image]

VWthefirst, 2024-06-23:
: ) schon klar das in meinem Fall das Problem vor dem Monitor sitzt. Hast du eine Idee woran es liegen könnte. Ich habe die Befehle per API und per Hand geschickt. Eigentlich ist da ja kein "man in the middle", der irgendetwas verändern könnte, oder?

knickohr, 2024-06-23:
Nein, das Problem sitzt weder davor noch dahinter oder sogar zwischen PC und Monitor 🤪

Das Problem sitzt in dieser kleinen Kiste und nennt sich Firmware. Nimm mal eine neue DEV 😇

VWthefirst, 2024-06-24:
Bombe !! Danke für deine Hilfe.

DanielR92, 2024-06-29:
Hat sich hier erledigt. Closed.

closed 2024-06-29 by DanielR92