lumapu/ahoy #1685: [Bug] Abfrage der Daten mit MC in MicroPython funktioniert nicht mehr mit Version 0.8.127

RF-Signal, 2024-06-23:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.127

### Github Hash

8cd364

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

HMS-1600
HMS-2000
CMT2300A radio enable
Display = SH1106
Esp32-wroom32 (ESP32-D0WD-V3)
Interval [s] = 20

### Debug Serial Log output

```shell
nicht relevant
```

### Error description

Ich benütze die Ahoy-DTU um zwei HMS Wechselrichter auszulesen und frage Daten von der DTU mit einem PicoW (RP2040) mit MicroPython ab.

Mit der Firmware Version 0.8.83 funktioniert das mit diesem Befehl problemlos:
**dtu = requests.get("http://192.168.2.119/api/inverter/id/0")**

Mit der Firmware Version 0.8.127 funktioniert dies nicht mehr. Ich bekomme keine Daten.

dknubben, 2024-06-23:
Hast Du die URL http://192.168.2.119/api/inverter/id/0 mal direkt in einem Browser aufgerufen?
Kommen dann JSON-Daten an?
Ich kann das Problem nicht nachvollziehen (0.8.127, allerdings HM-Inverter), evtl. gab es aber Formatänderungen bei den Daten, die in Deiner MicroPython-Umsetzung nicht berücksichtigt sind.

dknubben, 2024-06-27:
Ich kann Dir in dieser Konstellation sicher nicht weiterhelfen. Das könnten allenfalls die Entwickler dieses Projekts,
wobei Deine MicroPython-Lösung nicht in den "Support-Bereich" dieses Projekts fällt.

Mit 0.8.94 gab es diese Änderung:
"switched AsyncWebServer library"
Evtl. ist da die Ursache zu suchen.
Ansonsten wäre nach meiner Ansicht der richtige Ansatz, zuerst die Header der Antwort auf die GET-Anfrage anzusehen (Konsole im Browser). Sind hier Unterschiede bei den Versionen im Header festzustellen, z.B. Format, Codierung etc?

lumapu, 2024-07-07:
es wäre echt hilfreich die Header sehen zu können. Man kann scheinbar auch MicroPython in Docker ausführen, evtl. kommt man dann auch auch noch mit Wireshark dran um alles mitzuloggen. Leider fehlt mir gerade die Zeit dies zu tun.

lumapu, 2024-08-11:
I evaluiere gerade einen neuen Webserver, evtl. könntest du mal die angehängte Firmware testen, ob die wieder mit MC funktioniert. (https://github.com/mathieucarbou/ESPAsyncWebServer @ ^3.1.3)

[240811_ahoy_dev_esp32_0.8.135.zip](https://github.com/user-attachments/files/16576047/240811_ahoy_dev_esp32_0.8.135.zip)

RF-Signal, 2024-08-12:
Hallo,
ich bin noch 2 Wochen im Urlaub.
Wenn ich zurück bin teste ich das gerne.
Bernd 

Am 11. August 2024, 16:26, schrieb Lukas Pusch:

I evaluiere gerade einen neuen Webserver, evtl. könntest du mal die angehängte Firmware testen, ob die wieder mit MC funktioniert. (https://github.com/mathieucarbou/ESPAsyncWebServer @ ^3.1.3)

240811_ahoy_dev_esp32_0.8.135.zip

RealNBB, 2024-08-16:
Hi,

gerade die verlinkte Version installiert und am beobachten. Bisher keine Probleme.

[240811_ahoy_dev_esp32_0.8.135.zip](https://github.com/user-attachments/files/16576047/240811_ahoy_dev_esp32_0.8.135.zip)

Ich gebe morgen mal bescheid, ob alles weiterhin stabil funktioniert.

RF-Signal, 2024-08-27:
Hallo Lukas,

 

bin wieder zurück und habe die Firmware 0.8.135 geladen.

 

Ja der der Zugriff mit dem MC funktioniert jetzt wieder. (SUPER)

 

Allerdings habe ich jetzt ein anderes Problem:

Nach einem Reset kann ich vom PC aus mit dem Browser nicht mehr auf die DTU zugreifen!

 

 

 

 

 

 

 

 

 

 

IP Adr. stimmt.

 

http://192.168.2.119/

 

 

 

 

 

 

 

Jetzt geht gar nichts mehr.

 

Viele Grüße

Bernd

Gubi2023, 2024-08-27:
die .135 hat einen bug, nach  Änderungen schmiert die DTU ab.  Versuch doch mal die .140 release?

stefan123t, 2024-10-30:
@lumapu mathieucarbou hat aktuell v3.3.21 der ESPAsyncWebServer Bibliothek online gestellt.
Es gab ein [Heap / Memory Leak](https://github.com/tbnobody/OpenDTU/issues/2314) Problem beim Verwenden der alten API Aufrufe. 
Der Fix findet sich dann m.W. hier: https://github.com/tbnobody/OpenDTU/pull/2320

stefan123t, 2024-11-03:
@RF-Signal Bernd hast Du mal geklärt ob es vielleicht an den Headern liegt / lag ?

Bei mir erfolgt der Request mit den folgenden Request Headern:
```
GET /api/inverter/id/0 HTTP/1.1
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9
Cache-Control: no-cache
Connection: keep-alive
Host: opendtu-fusion
Pragma: no-cache
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36
```

die Antwort kommt dann mit
```
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: content-type
Connection: close
Accept-Ranges: none
Content-Length: 475
Content-Type: application/json
```

Ich hätte also erwartet dass die Anfrage auch mit `Accept: application/json` gestellt wird.