lumapu/ahoy #1686: [Bug]

lowsax, 2024-06-24:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.127

### Github Hash

8cdf364

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

kein MQTT, kein Seriell
2 Wandler, 3 Module
10 Sekunden

### Debug Serial Log output

```shell
{"version":11,"wifi":{"ssid":"XYZ.net","pwd":"","ap_pwd":"esp_8266","hidd":false,"dev":"PV","adm":"","prot_mask":61,"dark":true,"reb":false,"region":0,"timezone":1,"ip":"","mask":"","dns1":"","dns2":"","gtwy":""},"nrf":{"cs":15,"ce":2,"irq":0,"sclk":0,"mosi":0,"miso":0,"en":true},"ntp":{"addr":"192.168.53.254","port":123,"intvl":720},"sun":{"lat":50.0000000,"lon":9.0000000,"offs":900,"offsEve":900},"serial":{"show":false,"debug":false,"prv":false,"trc":false,"mqtt":false},"mqtt":{"broker":"","port":1883,"clientId":"","user":"","pwd":"","topic":"inverter","json":false,"intvl":0,"retain":true},"led":{"0":255,"1":255,"2":255,"act_high":true,"lum":127},"plugin":{"disp":{"type":1,"pwrSafe":false,"screenSaver":1,"graph_ratio":0,"graph_size":2,"rotation":2,"contrast":60,"data":4,"clock":5,"cs":255,"reset":255,"busy":255,"dc":255,"pirPin":255},"cst_lnk":"","cst_lnk_txt":""},"inst":{"intvl":10,"rstMidNight":true,"rstNotAvail":false,"rstComStop":true,"rstComStart":false,"strtWthtTime":false,"rstMaxMidNight":false,"rdGrid":true,"iv":[{"en":true,"name":"Garage","sn":12345,"note":"6789","freq":0,"pa":255,"dis":false,"yield":[0,0,0,0,0,0],"pwr":[385,385,0,0,0,0],"chName":["Strasse","Garten","","","",""]},{"en":true,"name":"Laube","sn":9876,"note":"154321","freq":0,"pa":255,"dis":false,"yield":[0,0,0,0,0,0],"pwr":[385,0,0,0,0,0],"chName":["1","","","","",""]}]}}
```

### Error description

Der Verlauf (beide Diagramme) wird gelegentlich zurückgesetzt und der bisherige Verlauf geht verloren.
Ich vermute, dass dieser Fall eintritt, wenn die Verbindung zu den Wandlern unterbrochen wird. Ich kann aber auch einen (ungewollten) Reset nicht auschließen

[image]

Gubi2023, 2024-06-24:
mit dem Problem kämpfe ich schon seit einem Jahr, dass sich die DTU urplötzlich neu startet (ersichtlich im Weblog, das dann 10 min. keine Daten mehr liefert und danach die Uhrzeit um 2 Std nachgeht). Leider noch keine Abhilfe gefunden.

Dirk-Ahoy, 2024-06-26:
Schmeisst mal den 8266 weg :)  

[image]

Gubi2023, 2024-06-26:
schön wärs, ich hab schon seit Monaten ESP32 und immer noch das gleiche Problem.
Heute mind. 3 Reboots ohne ersichtlichen Grund (V0.8.127)

technics42, 2024-06-27:
Using V0.8.101 on ESP8266 now for 36 days without any reboot - but only by reducing to show webpages in any browser.  (Only using webpages in very rare cases)
Handling 2 inverters with 10sec interval.
Requesting Rest-Api also with 10sec interval - alternating calls for inverters. (Unfortunately one cannot get all inverter data by just one single call.)

Reboots are caused when using/refreshing webpages too frequently. They are also just updated by using the same Rest-Api - number of calls increases by amount of inverters and interval of refreshing data is the same as interval of polling data from inverter.
One could improve stability by reducing webserver-calls, e.g. adding Rest-Api call for all inverter-data, using different polling intervals or even better by using websockets.

(To be honest, support of ESP8266 is the unique selling point for this project!)

lowsax, 2024-06-28:
Mit den vorangegangenen Versionen, sowohl release wir dev, hatte ich das nicht.
Gestern habe ich abends das erste mal die Seite geöffnet und trotzdem war vormittags ein Datenverlust. Am refresh kann es also eigentlich nicht liegen.

steve-zero, 2024-07-21:
Die aktuelle 0.8.129 ist bei mir auch noch immer instabil.
Bei mehrfachem Wechsel der Bereiche (Daten/Verlauf/Einstellungen, etc.) auf dem Webinterface oder Browser-Reloads crasht der 8266 immer wieder.
Besonders der Tab "Verlauf" ist fragil und kämpft mit Ladeverzögerungen, die dann oft zum Crash führen.

Gubi2023, 2024-07-21:
benutzt du MqTT? 
@lumapu hat festgestellt: MqTT läuft voll und bekommt die Pakete nicht abgeschickt, Ahoy bricht dann zusammen, da kein Speicher mehr da ist

steve-zero, 2024-07-21:
MqTT ist aktiv.
Ich bekomme mit der 129 den ESP aber auch innerhalb von weniger als einer Minute zum crashen, wenn ich mit dem Browser viel Last erzeuge.
Wenn ich ihn komplett in Ruhe lasse, rebootet er trotzdem alle 1-2 Tage. Das könnte dann vom MqTT kommen.

Bin jetzt auf die 121 zurück. Abgesehen von der offiziellen 0.8.83 (die noch keinen Tagesverlauf kann), ist das die letzte halbwegs stabile Version für mich.

Gubi2023, 2024-07-21:
d.h. die 121 läuft bei dir auch mehrere Tage durch?

steve-zero, 2024-07-21:
Von Ende Mai bis Anfang Juni ist mir da nichts negatives im Gedächtnis. Wobei ich da auch nicht darauf geachtet hatte.
Mit der 123 ging der Ärger subjektiv los. Und nachdem auch die 129 keine Besserung brachte, bin ich in den letzten Tagen rückwärts durch mein Archiv und die 121 war die letzte, die zumindest nicht direkt am Webinterface hängen blieb.
Seit gestern läuft sie jedenfalls.
Ich hoffe, dass sie nun durchläuft, aber ich weiß es nicht mit Sicherheit. Sorry.

steve-zero, 2024-07-21:
Hmm. Die 121 hat sich nach über 1 Tag eben aufgehängt, als ich die Live-Daten abfragen wollte. Historie ging aber vorher problemlos.
Könnte der MqTT Bug sein. Ich schalte MqTT mal probehalber ab, da ich derzeit eh keine Auswerteung laufen lasse.

steve-zero, 2024-07-28:
Nach Abschaltung von MqTT ist die 121 vollkommen stabil. Hab nun eine Woche Uptime.
Jede Version danach crasht aber bei Nutzung des Webinterfaces.
Ich würde mal sagen, dass das Update der AsyncWebserver library auf 3.2.2 auf dem 8266 nicht stabil läuft.

stefan123t, 2024-11-04:
@lowsax @steve-zero @Gubi2023 ich glaube das Problem mit den sporadischen Reboots ist auch in der v0.8.152 noch nicht ausreichend analysiert / gefixed ?

Gubi2023, 2024-11-04:
mein ESP8266 läuft mit der rel 8.140 seit 20 Tagen (allerdings **ohne MqTT**), weiss deshalb nicht, ob es mit der dev auch gehen würde. Gehe aber davon aus, dass es mit MqTT nicht geht, zumal mein ESP32 mit aktueller Dev regelmässig 2-3 mal täglich bootet...

steve-zero, 2024-11-05:
Es dürfte letzlich an der dynamischen Speicherbelegung liegen, die unter bestimmten Bedingungen (MQTT, Verlauf, Konsole) zu viel wird (vor allem die Zusammenstellung der Strings fürs Webinterface).
Da müsste man eine genaue Abschätzung des worst-case Speicherbedarfs machen und dann limitieren, welche Features überhaupt in einen ESP8266 Build noch rein dürfen.
Am sinnvollsten wäre es vermutlich auf den Verlauf und ein grafisches Display zu verzichten, da auf dem kleinen ESP das Polling und die Weiterleitung nach MQTT der Hauptzweck sein dürfte. Mehr dann halt nur auf einem ESP32.