lumapu/ahoy #1688: Übersicht auf I-Phone: Text unvollständig[Bug]

Gubi2023, 2024-06-25:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.127

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

V0.8.127 auf Handy

### Error description

in der Übersicht auf dem Handy fehlt das Wort "verfügbar" siehe Screenshot
[image]

knickohr, 2024-06-25:
Tatsächlich !  😲

[image]

Ist wohl schon vorher rein gekommen, meins ist 125. ist noch niemanden aufgefallen 🫣

Gubi2023, 2024-06-25:
tja, die .125 und .126  hab ich übersprungen.

knickohr, 2024-06-25:
Wobei es in englisch nicht mal so schlimm ist und sogar grammatikalisch richtig 😉

Aber deutsch geht’s ja gar nicht 😲

Gubi2023, 2024-06-25:
da müsste man dann das "ist" auch noch weglassen, dann wäre es wieder gleich. Wie ist denn der engl. Laptop-Text?

knickohr, 2024-06-25:
Das kleinste iPad reicht schon aus 😇

[image]

geronet1, 2024-06-28:
Dann muss man nur die Übersetzung anpassen, sonst sieht das nämlich ätzend aus:
https://github.com/lumapu/ahoy/pull/1663#issuecomment-2156341049

Gubi2023, 2024-08-05:
Bug ist in V 0.8.130 immer noch drin....

Gubi2023, 2025-03-07:
Fehler ist im Release 00.8.154 wieder drin!

Gubi2023, 2025-03-11:
mit Release 0.8.155 hoffentlich entgültig beseitigt..

closed 2025-03-11 by Gubi2023