lumapu/ahoy #1689: Aktualisierung mit V0.8.127 zu langsam?

Gubi2023, 2024-06-25:
mit der V0.8.127 werden meine Gesamtwere teilweise nur verzögert aufaddiert. Das führt im Verlauf zu einem unschönen Sägezahnmuster (Aktualisierungsintervall 20 Sec.) Siehe Screenshots
[image]

[image]

technics42, 2024-06-25:
In your screenshot DC-power of inverter looks good - sum of panel's DC-power.
As far as I know, AC-power is read directly from inverter.
What about your power limit of 150%?
Are you using any kind of battery?

dknubben, 2024-06-25:
Kann das mit Flole zusammenhängen? Ich habe das Tool zwar auch bei 3 HM-600 verwendet, aber nur auf jeweils max. 133% (knapp die 800 W). Auf mehr würde ich nie gehen. Der Leistungsverlauf ist dabei völlig in Ordnung. Evtl. mal auf originalen Zustand zurücksetzen und neu prüfen.

Gubi2023, 2024-06-25:
wäre möglich. Habe jetzt auch gesehen, dass es bei höheren Wattzahlen richtig funktioniert. War also etwas vorschnell und mache deshalb den Issue wieder zu

closed 2024-06-25 by Gubi2023