lumapu/ahoy #1691: Einstellung Sendeleistung V0.8.**

DerBorzi, 2024-06-26:
Hallo zusammen,
habe heute meine DTU von 0.7.*** dev geupdatet auf 0.8.83, leider fehlt die Einstellmöglichkeit der Sendeleistung.
Ich bin leider etwas weit vom HM-600 entfernt, mit 0.7.36 kann ich die Sendeleistung auf MAX stellen, dann klappts auch mit der Kommunikation problemlos, die anderen Einstellungen bleiben alle gleich.

Geht das noch anders über Umwege?

VG
Andi

## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2024-06-26:
Ist es nicht, schau mal bei den Invertersettings (das Zahnrad).

Ansonsten, Du hast einen 8266. Funktioniert überhaupt die Verbindung zum nRF-Modul ?  Wenn nicht, vielleicht mal D3 und D4 vertauschen.

DerBorzi, 2024-06-26:
Siehe oben, Einstellung gefunden, klappt trotzdem nicht. D3 (IRQ) und D4 (CE) sollte passen (hatte vorher schon 0.7x wo die Änderung war).
Status vom Funkmodul ist "grün", also connected, Interrupt Pin ist auch "grün".

Nach Rückflash auf 0.7.36 klappt die Verbindung auf Anhieb.

Gubi2023, 2024-06-26:
was spricht das webserial?

rejoe2, 2024-06-26:
Eine andere Möglichkeit: #1434 (bzw. das dahinter angesprochene `startFastWrite()`.

Zum Wiederfinden - es gibt noch eine Reihe weiterer Issues mit (vielleicht) ähnlichen Symptomen: #1642 #1489. 

Bitte erst mal 
- den nRF checken, ob der "genuine" ist;
- ein anderes Netzteil verwenden (da der nRF etwas mehr Strom braucht, um den lna aktiv zu halten);
- den Power-Level reduzieren. Wir haben User, die gute 80m mit weniger wie HIGH überbrücken können (bei näherungsweise 100% success ratio).

DerBorzi, 2024-06-27:
Guten Morgen,

ich weiß nicht so genau wie man lange Logs lesbar hier einfügt, daher als Anhang, die Namen sollten selbsterklärend sein. 

Zusammenfassung:
V0.7.36 HIGH: Es klappt nicht jede Kommunikation
[2024-06-27_V0.7.36_LOG_HIGH.txt](https://github.com/user-attachments/files/16008548/2024-06-27_V0.7.36_LOG_HIGH.txt)
[image]

V0.7.36 MAX: Kommunikation klappt recht zuverlässig.
[2024-06-27_V0.7.36_LOG_MAX.txt](https://github.com/user-attachments/files/16008554/2024-06-27_V0.7.36_LOG_MAX.txt)

PIN-Settings siehe Screenshot.

In beiden Fällen aber auf Anhieb nach dem (Neu-)Start.

V0.8.83:
Gleiche Hardware, gleiches Setup, keine Kommunikation egal mit welcher Sendeleistung.
[2024-06-27_V0.8.83_LOG_MAX.txt](https://github.com/user-attachments/files/16008555/2024-06-27_V0.8.83_LOG_MAX.txt)
![2024-06-27_PinConfig_0 8 83](https://github.com/lumapu/ahoy/assets/131904486/3b0e24ec-2678-4a07-941d-2a9f
[image]
a26a133e)

Netzteil würde ich ausschließen, da es mit V0.7 fiunktioniert, es ist ein 2,5A RasPI Netzteil.

Was meinst du mit NRF generic?

Danke euch!

VG
Andi

Gubi2023, 2024-06-27:
versuch doch mal spasseshalber eine neuere dev-Version, wie es da aussieht. https://fw.ahoydtu.de/fw/

Gubi2023, 2024-06-27:
tja für mich als Laie sieht es so aus, als ob das NRF-Modul  mit der neuen Version nicht klarkommt (da keine Rx zurückkommen)
Vllcht ist es doch zu schwach auf der Brust... Das einfachste wäre. auf der alten Vesion zu bleiben und mittelfristig ne DTU mit ESP32 zu besorgen

knickohr, 2024-06-27:
Wie lange hast Du gewartet ?  Die 8-Versionen haben eine Heuristik und suchen sich den besten Kanal aus. Von den Zeiten her sehe ich das Du alle 5s abfragst.solltest also schon ca. 3 Minuten warten bevor Du aufgibst.

rejoe2, 2024-06-27:
Nochmal: Tausch probeweisen das Netzteil, es wäre nicht das erste Mal, dass sowas über der Zeit degradiert...

Die Logik ist an manchen Stellen anders, und V0.8.x braucht seit der genannten Änderung definitiv MEHR STROM!

Und mit "genuine" ist gemeint: Was ist das für ein nRF-Modul. UU. haben "fake"-Module (die "nscheinend tadellos funktionieren" ein Problem, wenn sie "richtig" burst mode machen sollen... Ggf. bitte mal ein Bild von dem Ding machen, das du verwendest.

DerBorzi, 2024-06-27:
> Wie lange hast Du gewartet ? Die 8-Versionen haben eine Heuristik und suchen sich den besten Kanal aus. Von den Zeiten her sehe ich das Du alle 5s abfragst.solltest also schon ca. 3 Minuten warten bevor Du aufgibst.

Krass, das war der entscheidende Hinweis, geduldig knapp 3 Minuten warten, dann klappts auch mit der Übertragung. 
Dennoch interessant: ein ESP32 mit CMT2300A und HMS-1800 findet den WR gleich nach StartUp. Lässt sich sicher auch irgendwie erklären.

Netzteiltausch hat sich somit erledigt, Funkmodul ist zwar nicht schuld aber dieses hier:
[image]
[image]

Danke an alle für die zeitnahe Unterstützung!

VG
Andi

rejoe2, 2024-06-27:
Sehe den angeblich vorhandenen C nicht....

Dann sollte es auch mit low gehen....

knickohr, 2024-06-27:
Ein HMS hat keine Heuristik 😉

DerBorzi, 2024-06-27:
C ist schon drin, unglückliches Foto:
[image]

Danke @knickohr für die Erklärung!

rejoe2, 2024-06-28:
OK, na dann... Es sollte aber trotzdem (mit aktuellen dev-Versionen) keine 3 Minuten (bei 5 sec Intervall) dauern, bis Ahoy was vom WR hört. U.a. durch die Änderung, die mehr Saft benötigt, kann (!) man häufig tx/rx-Raten bekommen, die nahe dem liegen, was mit CMT geht (da werden die Frequenzen anders gewählt, und die DTU "weiß" sicher, wo sie den WR findet, solange keine 2. DTU (oder ein reboot oä.) im Spiel ist).

Generell sind Dupont-Kabel halt kein Ersatz für ordentliche Platinen, und MAX ist afaik in D mit dieser Art Modul illegal. Besorge also lieber ordentliche Hardware (geschirmtes nRF, Platine, ggf. irgendwas ab ESP32)...

DerBorzi, 2024-06-30:
Nach deinem Hinweis habe ich die aktuelle dev installiert, hier wird der WR tatsächlich wieder schneller gefunden, quasi sofort.

kiu77, 2024-07-05:
Das nRF-Modul scheint mir auch kein Originales.

closed 2024-08-16 by lumapu