lumapu/ahoy #1693: Werte aktualisieren sich nicht mehr

beyerste, 2024-06-27:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nichts im Setup angepasst.
Nur den Inverter eingetragen.
Bis vor 2 Tagen hat noch alles ohne Probleme funktioniert.

### Debug Serial Log output

```shell
.8.83 - 5ebfe5a
---------------
15:18:02.376 I: (#0) request timeout: 400ms
15:18:02.377 I: (#0) TX 27 CH40, 15 ret. | 15  *  *  *  * 87 19 56 08 80 0b 00 66 7d 82 aa 00 00 00 00 00 00 00 00 aa 98 8c 
15:18:02.781 I: (#0) request timeout: 400ms
15:18:02.781 -----
15:18:02.781 I: com loop duration: 1655ms
15:18:02.782 -----
webSerial, buffer overflow!

15:18:06.130 I: Stellplatz/ch0/Efficiency: 95.000 %
15:18:06.130 I: Stellplatz/ch0/MaxPower: 68.400 W
15:18:06.131 I: 
02 I: Stellplatz/ch1/P_DC: 15.600 W
15:18:06.002 I: Stellplatz/ch1/YieldDay: 1009.000 Wh
15:18:06.003 I: Stellplatz/ch1/YieldTotal: 516.382 kWh
15:18:06.014 I: Stellplatz/ch1/Irradiation: 3.900 %
15:18:06.015 I: Stellplatz/ch1/MaxPower: 15.600 W
15:18:06.015 I: Stellplatz/ch2/U_DC: 28.200 V
15:18:06.016 I: Stellplatz/ch2/I_DC: 0.700 A
15:18:06.026 I: Stellplatz/ch2/P_DC: 19.600 W
15:18:06.027 I: Stellplatz/ch2/YieldDay: 1143.000 Wh
15:18:06.028 I: Stellplatz/ch2/YieldTotal: 518.127 kWh
15:18:06.038 I: Stellplatz/ch2/Irradiation: 4.900 %
15:18:06.039 I: Stellplatz/ch2/MaxPower: 19.600 W
15:18:06.040 I: Stellplatz/ch3/U_DC: 27.900 V
15:18:06.040 I: Stellplatz/ch3/I_DC: 0.740 A
15:18:06.051 I: Stellplatz/ch3/P_DC: 20.700 W
15:18:06.052 I: Stellplatz/ch3/YieldDay: 1228.000 Wh
15:18:06.052 I: Stellplatz/ch3/YieldTotal: 513.887 kWh
15:18:06.053 I: Stellplatz/ch3/Irradiation: 5.175 %
15:18:06.064 I: Stellplatz/ch3/MaxPower: 20.700 W
15:18:06.064 I: Stellplatz/ch4/U_DC: 27.900 V
15:18:06.065 I: Stellplatz/ch4/I_DC: 0.580 A
15:18:06.065 I: Stellplatz/ch4/P_DC: 16.100 W
15:18:06.076 I: Stellplatz/ch4/YieldDay: 752.000 Wh
15:18:06.077 I: Stellplatz/ch4/YieldTotal: 439.026 kWh
15:18:06.077 I: Stellplatz/ch4/Irradiation: 4.025 %
15:18:06.078 I: Stellplatz/ch4/MaxPower: 16.100 W
15:18:06.089 I: Stellplatz/ch0/U_AC: 237.400 V
15:18:06.089 I: Stellplatz/ch0/I_AC: 0.290 A
15:18:06.090 I: Stellplatz/ch0/P_AC: 68.400 W
15:18:06.090 I: Stellplatz/ch0/Q_AC: 21.900 var
15:18:06.101 I: Stellplatz/ch0/F_AC: 49.970 Hz
15:18:06.102 I: Stellplatz/ch0/PF_AC: 0.952 
15:18:06.102 I: Stellplatz/ch0/Temp: 48.100 °C
15:18:06.103 I: Stellplatz/ch0/ALARM_MES_ID: 1.000 
15:18:06.114 I: Stellplatz/ch0/YieldDay: 4132.000 Wh
15:18:06.114 I: Stellplatz/ch0/YieldTotal: 1987.422 kWh
15:18:06.115 I: webSerial, buffer overflow!

15:18:11.130 I: Stellplatz/ch0/Efficiency: 95.000 %
15:18:11.131 I: Stellplatz/ch0/MaxPower: 68.400 W
15:18:11.131 I: 
02 I: Stellplatz/ch1/P_DC: 15.600 W
15:18:11.003 I: Stellplatz/ch1/YieldDay: 1009.000 Wh
15:18:11.003 I: Stellplatz/ch1/YieldTotal: 516.382 kWh
15:18:11.014 I: Stellplatz/ch1/Irradiation: 3.900 %
15:18:11.015 I: Stellplatz/ch1/MaxPower: 15.600 W
15:18:11.015 I: Stellplatz/ch2/U_DC: 28.200 V
15:18:11.016 I: Stellplatz/ch2/I_DC: 0.700 A
15:18:11.027 I: Stellplatz/ch2/P_DC: 19.600 W
15:18:11.027 I: Stellplatz/ch2/YieldDay: 1143.000 Wh
15:18:11.028 I: Stellplatz/ch2/YieldTotal: 518.127 kWh
15:18:11.039 I: Stellplatz/ch2/Irradiation: 4.900 %
15:18:11.039 I: Stellplatz/ch2/MaxPower: 19.600 W
15:18:11.040 I: Stellplatz/ch3/U_DC: 27.900 V
15:18:11.040 I: Stellplatz/ch3/I_DC: 0.740 A
15:18:11.051 I: Stellplatz/ch3/P_DC: 20.700 W
15:18:11.052 I: Stellplatz/ch3/YieldDay: 1228.000 Wh
15:18:11.052 I: Stellplatz/ch3/YieldTotal: 513.887 kWh
15:18:11.053 I: Stellplatz/ch3/Irradiation: 5.175 %
15:18:11.064 I: Stellplatz/ch3/MaxPower: 20.700 W
15:18:11.064 I: Stellplatz/ch4/U_DC: 27.900 V
15:18:11.065 I: Stellplatz/ch4/I_DC: 0.580 A
15:18:11.066 I: Stellplatz/ch4/P_DC: 16.100 W
15:18:11.076 I: Stellplatz/ch4/YieldDay: 752.000 Wh
15:18:11.077 I: Stellplatz/ch4/YieldTotal: 439.026 kWh
15:18:11.078 I: Stellplatz/ch4/Irradiation: 4.025 %
15:18:11.078 I: Stellplatz/ch4/MaxPower: 16.100 W
15:18:11.089 I: Stellplatz/ch0/U_AC: 237.400 V
15:18:11.089 I: Stellplatz/ch0/I_AC: 0.290 A
15:18:11.090 I: Stellplatz/ch0/P_AC: 68.400 W
15:18:11.091 I: Stellplatz/ch0/Q_AC: 21.900 var
15:18:11.101 I: Stellplatz/ch0/F_AC: 49.970 Hz
15:18:11.102 I: Stellplatz/ch0/PF_AC: 0.952 
15:18:11.103 I: Stellplatz/ch0/Temp: 48.100 °C
15:18:11.103 I: Stellplatz/ch0/ALARM_MES_ID: 1.000 
15:18:11.114 I: Stellplatz/ch0/YieldDay: 4132.000 Wh
15:18:11.114 I: Stellplatz/ch0/YieldTotal: 1987.422 kWh
15:18:11.115 I: webSerial, buffer overflow!

15:18:16.130 I: Stellplatz/ch0/Efficiency: 95.000 %
15:18:16.130 I: Stellplatz/ch0/MaxPower: 68.400 W
15:18:16.131 I: 
15:18:16.132 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 34, s: 1, f: 7, n: 26 | p: 2
15:18:16.134 I: (#0) TX 27 CH61, 15 ret. | 15  *  *  *  * 87 19 56 08 80 0b 00 66 7d 82 b8 00 00 00 00 00 00 00 00 0a 4c ea 
15:18:16.548 I: (#0) request timeout: 400ms
15:18:16.549 I: (#0) TX 27 CH75, 15 ret. | 15  *  *  *  * 87 19 56 08 80 0b 00 66 7d 82 b8 00 00 00 00 00 00 00 00 0a 4c ea 
15:18:16.953 I: (#0) request timeout: 400ms
15:18:16.953 I: (#0) TX 27 CH03, 15 ret. | 15  *  *  *  * 87 19 56 08 80 0b 00 66 7d 82 b8 00 00 00 00 00 00 00 00 0a 4c ea 
 4.900 %
15:18:16.039 I: Stellplatz/ch2/MaxPower: 19.600 W
15:18:16.040 I: Stellplatz/ch3/U_DC: 27.900 V
15:18:16.040 I: Stellplatz/ch3/I_DC: 0.740 A
15:18:16.051 I: Stellplatz/ch3/P_DC: 20.700 W
15:18:16.052 I: Stellplatz/ch3/YieldDay: 1228.000 Wh
15:18:16.052 I: Stellplatz/ch3/YieldTotal: 513.887 kWh
15:18:16.053 I: Stellplatz/ch3/Irradiation: 5.175 %
15:18:16.064 I: Stellplatz/ch3/MaxPower: 20.700 W
15:18:16.064 I: Stellplatz/ch4/U_DC: 27.900 V
15:18:16.065 I: Stellplatz/ch4/I_DC: 0.580 A
15:18:16.065 I: Stellplatz/ch4/P_DC: 16.100 W
15:18:16.076 I: Stellplatz/ch4/YieldDay: 752.000 Wh
15:18:16.077 I: Stellplatz/ch4/YieldTotal: 439.026 kWh
15:18:16.077 I: Stellplatz/ch4/Irradiation: 4.025 %
15:18:16.078 I: Stellplatz/ch4/MaxPower: 16.100 W
15:18:16.089 I: Stellplatz/ch0/U_AC: 237.400 V
15:18:16.089 I: Stellplatz/ch0/I_AC: 0.290 A
15:18:16.090 I: Stellplatz/ch0/P_AC: 68.400 W
15:18:16.091 I: Stellplatz/ch0/Q_AC: 21.900 var
15:18:16.101 I: Stellplatz/ch0/F_AC: 49.970 Hz
15:18:16.102 I: Stellplatz/ch0/PF_AC: 0.952 
15:18:16.102 I: Stellplatz/ch0/Temp: 48.100 °C
15:18:16.103 I: Stellplatz/ch0/ALARM_MES_ID: 1.000 
15:18:16.114 I: Stellplatz/ch0/YieldDay: 4132.000 Wh
15:18:16.114 I: Stellplatz/ch0/YieldTotal: 1987.422 kWh
15:18:16.115 I: 15:18:17.357 I: (#0) request timeout: 400ms
15:18:17.358 I: (#0) TX 27 CH23, 15 ret. | 15  *  *  *  * 87 19 56 08 80 0b 00 66 7d 82 b9 00 00 00 00 00 00 00 00 9a 41 76 
15:18:17.762 I: (#0) request timeout: 400ms
15:18:17.762 -----
15:18:17.762 I: com loop duration: 1632ms
15:18:17.763 -----
webSerial, buffer overflow!

15:18:21.130 I: Stellplatz/ch0/Efficiency: 95.000 %
15:18:21.130 I: Stellplatz/ch0/MaxPower: 68.400 W
15:18:21.131 I: 
02 I: Stellplatz/ch1/P_DC: 15.600 W
15:18:21.002 I: Stellplatz/ch1/YieldDay: 1009.000 Wh
15:18:21.003 I: Stellplatz/ch1/YieldTotal: 516.382 kWh
```

### Error description

Bis vor 2 Tagen hat noch alles ohne Probleme funktioniert.
Jetzt werden nur noch sporadisch Livedaten angezeigt.
Meist kommt ein Satz Daten und danach wird nicht mehr aktualisiert.
Im WebSerial kommen dann Fehlermeldugngen "webSerial, buffer overflow!".

knickohr, 2024-06-27:
Ich sehe kein RX und auch kein TX. Sicher das die Kommunikation zu den WR noch funktioniert ?

beyerste, 2024-06-27:
Danke für den Hinweis.
Ich dachte, dass die Kommunikation wegen der Fehlermeldung abbricht.
Das Problem kam ja plötzlich, nachdem es funktioniert hatte.
Eventuell liegt es aber gar nicht an der Software oder dem WR, es könnte sein, dass die Funkverbindung unterbrochen wird. 
Da es mal geht und mal nicht, wäre eventuell die Vegetation, die zwischen Sender und Empfänger ist schuld an der Unterbrechnung. Ich werde das einmal testen.

Gubi2023, 2024-06-27:
und probehalber mal Kommunikation runterdrehen, dann gibts kein Buffer overflow mehr. Angezeigte Einstellungen reichen normalerweise...

[image]

beyerste, 2024-06-28:
Danke für die Unterstützung.

closed 2024-06-28 by beyerste