lumapu/ahoy #1694: AhoyDTU findet WR nicht mehr

RufusRed80, 2024-06-28:
## Hardware
  * [x] ESP8266

### nRF24L01+ Module
* [x] NRF24L01

### Antenna:
* [x] external antenna (SMA)

### Version / Git SHA: 
Version: 0.8.83

### Build & Flash Method:
* [x] AhoyDTU Webinstaller

Moin zusammen,

seit ein paar Tagen mag meine AhoyDTU nicht mehr mit dem WR kommunizieren. Strom wird aber produziert (Shelly meldet weiterhin eine Stromproduktion). 

NRF24L01 | connected
250 kbps
80703605

MqTT
connected
11580
2

AhoyDTU habe ich schon neu gestartet, den WR (Hoymiles 1500) auch. Wie kann ich das Problem eingrenzen?

LG Markus

DanielR92, 2024-06-28:
Handelt es sich um ein selbst gebaute Hardware? 
Wenn ja, mal alle verbindungen prüfen. 

Jumperkabel sind hier mal näher zu betrachten.

Loetnase, 2024-06-29:
Bekommt deine DTU eine Uhrzeit?
Ohne Zeit geht nichts, ansonsten mal mit Browsertime synchronisieren

Sme1d, 2024-07-14:
Check mal unter "System" den Status von
 - RF24L01	=> connected
 - Interrupt Pin working	 => true
 
Fals dem nicht so ist, würde ich deine (Löt-/Steck-) Verbindungen und deine Pin-Belegung checken. Bei mir war das der Fehler.

RufusRed80, 2024-07-14:
Moin,
Problem ist erledigt bzw. gefunden. Die Antenne hatte scheinbar einen Bruch in der Leitung?! Mit einer neuen Antenne funktioniert das System wieder wie gewünscht. DANKE EUCH :)

closed 2024-07-14 by RufusRed80