lumapu/ahoy #1695: GRID wird nicht erkannt

chr-braun, 2024-06-28:
## Hardware
  * [ ] ESP8266
  * [ x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.83
Github Hash: 5ebfe5a

### Build & Flash Method:
* [ X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

GRID INFO ROHDATEN: 00 01 80 02 00 D8 46 C3 46 C3 FF FF F1 ED 
GRID INFO ROHDATEN: 00 01 80 02 00 5E 43 C6 43 C6 FF FF F1 ED

Gubi2023, 2024-06-28:
Versuch doch mal eine neuere dev-Version, wie es da aussieht. https://fw.ahoydtu.de/fw/

chr-braun, 2024-06-29:
Scheint damit zu laufen, danke dir

closed 2024-08-16 by lumapu