lumapu/ahoy #1696: DTU Sniffer

VW181-Cologne, 2024-07-04:
Hallo zusammen.

Ich habe für einen Freund euer Modul zum Einsatz gebracht.
Funktioniert prima.
Während der Implementationsphase hatte ich mich etwas genauer mit dem Thema beschäftigt.
Leider sind alle meine Informationen futsch. Mist.

Nun mein Anliegen.
Ich möchte (experimentell) einen DTU-Sniffer aufbauen, mit dem ich nach und nach die Befehle für den Wechselricher bestimmen kann.
Also auslesen und auch schreiben in den Wechelrichter. z.B. Wechselrichter wenn möglich auf ein anderes Leistungniveau heben oder senken.
Könnt ihr mir Internetseiten nennen bei denen ich die nötigen Informationen ansehen kann.
Vieleicht gibt es ja schon ein Projekt das man mit etwas Ehrgeiz so aufbereiten kann wie euer Tool.
Nicht falsch verstehen, dies ist nicht als Ersatz oder Konkurenz zu verstehen, dafür wäre mir meine Zeit zu schade. 
Also ein Tool um die Wechselrichter zu verstehen und zu bearbeiten. 
Damit die so gewonnenen Informationen der Allgemeinheit zur Verfügung gestellt werden kann.

Gruß
Jörg

DanielR92, 2024-07-04:
Hi,

_Leider sind alle meine Informationen futsch. Mist._
Wer kein Backup macht, kein Mitleid. (Meine Meinung)

All dein Vorhaben, kann Ahoy und openDTU bereits. 
Was für ein "mehrgewinn" oder in deinem Satz formuliert '_Damit die so gewonnenen Informationen der Allgemeinheit zur Verfügung gestellt werden kann._', soll es der Allgemein helfen? 

Wenn du sowas programmieren kannst, dann schau doch einfach den Code von diesen zwei Projekten an. 
Wenn du die Wiki-Seite von diesem Projekt liest, findest du auch einige Quellen. 

Noch fragen?

stefan123t, 2024-07-05:
Hallo Jörg @VW181-Cologne schau mal in unser Dokumentation: 

**Hoymiles Format Description**
https://github.com/lumapu/ahoy/blob/main/doc/hoymiles-format-description.md

Dort haben wir das bisherige Vorgehen via NRF24 Sniffer / HackRF beschrieben, als auch die Testpunkte auf den DTU WLite & Pro Boards des Herstellers dokumentiert. Der Source Code des ersten NRF24 Sniffers sollte auch noch in irgendeinem der Verzeichnisse unter [/tools](https://github.com/lumapu/ahoy/tree/main/tools) rumliegen.

Analog dazu gab es auch schon Ansätze einen WR zu simulieren um die DTU auch Nachts mit einem Gesprächspartner zu testen.
https://github.com/lumapu/ahoy/issues/612#issuecomment-1399640379

VW181-Cologne, 2024-07-05:
Hallo 
und danke für die schnellen Antworten. 

An DanielR92

Daniel du hast recht. Kein Mitleid.
Ich habe natürlich ein Backup.
Es ist aber so das ich das Verzeichnis wo meine Infomationen liegen nur nicht finde, was genau so schlimm ist.

Welche zwei Projekte meinst du?

An stefan123t

Stefan danke für die Anstöße, ich habe euren Github ordner noch nicht durchgeschaut.

stefan123t, 2024-07-05:
@VW181-Cologne 

> Welche zwei Projekte meinst du?
* https://github.com/lumapu/ahoy/
* https://github.com/tbnobody/OpenDTU/

Wie gesagt schau mal unter tools, aber die besten Ergebnisse erhälts Du mit einem Logic Shrimp, z.B. Salea Logic 2 oder ein kostengünstiges Derivat aus dem fernen Osten. Damit kannst Du die beiden RX/TX am NRF24L01+ bzw. CMT2300A in Deiner Hoymiles DTU ganz genau mitlesen. Das mit dem Aufzeichnen per Funk hatten wir ganz am Anfang, das ist Mist.

Wenn Du die Kommandos sehen willst, die das [Hoymiles Protokoll](https://github.com/lumapu/ahoy/wiki/Protocol) hat, dann schau doch einfach mal ins Wiki, dort haben wir den aktuellsten Stand im Prinzip dokumentiert. Es sind nur ganz wenige Kommandos die noch nicht implementiert, wie z.B. Grid Profile Update, aber dafür gibt es dann offene Issues im Issue Tracker der beiden o.g. Projekte.

closed 2024-08-16 by lumapu