lumapu/ahoy #1698: Neustart der DTU, wenn Feldstärke zu nieder?[Bug]

Gubi2023, 2024-07-07:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.127

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nur NRF mit HM600

### Error description

Meine DTU startet sich immer noch sporadisch neu.
Meine Vermutung: Die Feldstärkenanzeige nimmt im Verlauf des Tages immer weiter an. Anscheinend startet sich die DTU neu, wenn die Feldstärke unter einen bestimmten Wert fällt. Siehe Sccreenshot.   
[image]

Gubi2023, 2024-07-07:
siehe auch https://github.com/lumapu/ahoy/issues/1651#issuecomment-2198212153

knickohr, 2024-07-07:
Läuft MQTT ?

Ja, da könnte ein Bug drin sein wenn der Broker nicht erreichbar ist. Das Problem ist teilweise gefixed, PR wird folgen.

Gubi2023, 2024-07-07:
Hab MqTT mal abgeschaltet, mal sehen wie es morgen aussieht...

Gubi2023, 2024-07-09:
ohne Mqtt scheint es zu gehen. DTU seit 2 Tagen ohne Reboot...

knickohr, 2024-07-09:
Tja, wie schon geschrieben, das Problem ist momentan nur teilweise gefixed. Es wird n eine der nächsten DEVs als PR kommen 😉

Gubi2023, 2024-11-03:
hatte heute wieder einen Reboot der DTU ohne die normalerweise auftretende 10-Min Unterbrechung, so wie auch im Screenshot am Anfang des Issues dargestellt. Auch hier wieder sehr schlechte WLAN-Verbindung in der Fritzbox
habe V0.8.152 geladen