lumapu/ahoy #1699: [Bug] WARNING: publisher/pubMqtt.h 422:76

brekemeier, 2024-07-08:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a8

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

null

### Debug Serial Log output

```shell
from app.cpp:7:
publisher/pubMqtt.h: In member function 'void PubMqtt<HMSYSTEM>::discoveryConfigLoop()':
publisher/pubMqtt.h:422:76: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  422 |                 constexpr static char* unitTotal[] = {"W", "kWh", "Wh", "W"};
      |                                                                            ^
publisher/pubMqtt.h:422:76: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
publisher/pubMqtt.h:422:76: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
publisher/pubMqtt.h:422:76: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
Compiling .pio\build\esp8266-de\lib8e6\ESP Async WebServer\WebRequest.cpp.o
In file included from app.h:38,
                 from main.cpp:7:
hm/Communication.h: In member function 'void Communication::innerLoop(const CommQueue<>::queue_s*)':
hm/Communication.h:120:39: warning: suggest parentheses around '&&' within '||' [-Wparentheses]
  120 |                     if(!mIsRetransmit && (q->cmd == AlarmData) || (q->cmd == GridOnProFilePara))
      |                        ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
In file included from app.h:38,
                 from app.cpp:7:
hm/Communication.h: In member function 'void Communication::innerLoop(const CommQueue<>::queue_s*)':
hm/Communication.h:120:39: warning: suggest parentheses around '&&' within '||' [-Wparentheses]
```

### Error description

1. Syntaxerror: constexpr static char* unitTotal[] = {"W"} all must (char*)"W", (char*)"kWh", (char*)"Wh", (char*)"W"  
2. Syntaxerror: hm/communication:120:39 =  "&&" = | |

knickohr, 2024-07-09:
Bitte auf eine neuere DEV gehen. Die 83 ist schon ein bißchen alt, da hat sich mittlerweile viel getan, auch wenn es noch keine neuere Stable Release gibt.

brekemeier, 2024-07-09:
War nur eine Info. Ich hoffe ich hab es richtig gemacht. Fehler sollen wir ja melden. Ich arbeite mit der 0.8.127.
Grüße aus Hameln, donate gibt's nächsten Monat.

knickohr, 2024-07-09:
👍

lumapu, 2024-07-11:
danke für den Hinweis - das ist in der dev bereits behoben, wie @knickohr angenommen hat.

brekemeier, 2024-07-11:
@lumapu dank nicht nötig. Ich kompiliere mit VSC/platformIO um auch die Adressen für Boot/Part/firm zu sehen. So habe ich auch ein Error in der u8g2 gefunden und gemeldet. Ich habe den Error mit einem Build-Befehl: -DCORE_DEBUG_LEVEL=0 abgestellt, da es nur eine Warnung war. Probleme habe ich nur mit der Mischung von C und C++ Files. C habe ich ich vor 40 Jahren gelernt (Informatiker), C++ seit V5.x. Ihr macht eine schöne Arbeit und freue mich, wenn ich euch den Error-Level melden kann. Donate folgt nächsten Monat. Falls es möglich ist, würde ich mich, für eine größeres Displayformt, freuen. Die SH1106 ist etwas zu klein, so könnte die Grafik besser aufgeteilt werden. Danke für die Rückmeldung.  Bitte bei meinen Bugs nur zur Kenntnis nehmen.

knickohr, 2024-07-12:
Wie wäre es mit einem SSD1309 ?  2,42“ aber halt leider auch nur 128x64 Pixel. Mehr Information würde nur auf ein ePaper (400x400 Pixel) passen, ist aber von den Abmessungen auch nur recht klein.

Größere Displays müßten erst programmiert werden.

brekemeier, 2024-07-12:
Hi,

 

ich bestell mir mal eins. Die Liste von U8g2 ist .

ja lang

closed 2024-08-16 by lumapu