lumapu/ahoy #1702: V0.8.128: Meldungsanzeige

Gubi2023, 2024-07-12:
Ist es Absicht, dass in der letzen Version der Meldungszähler auf 0 steht, obwohl ein Ereignis (Inverterstart) hinterlegt ist?

knickohr, 2024-07-12:
Ja, „Inverster Start“ ist kein Alarm 😉

Juergen2453, 2024-07-12:
Das stimmt so nicht ganz. Neueste Version 0.8.129

[image]

Gubi2023, 2024-07-12:
ok, in 129 ist der Zähler wieder auf 1.

closed 2024-07-12 by Gubi2023

knickohr, 2024-07-12:
Ich persönlich würde das nicht Alarm nennen sondern Event/Ereignis.

Juergen2453, 2024-07-12:
Da steht doch "Meldungen : 1" -> "Meldungen von ..." . Nirgendwo steht da was von "Alarm"

knickohr, 2024-07-12:
Ja, ab der 129

Gubi2023, 2024-07-12:
in der englischen Version heisst es schon immer Alarms, aber nicht in deutsch....

Juergen2453, 2024-07-12:
Es ist unbegreiflich. Es gab eine Frage, ich habe geantwortet mit Bild , es wurde bestätigt und das Thema geschlossen v. 129. Ob nicht richtig gelesen wird, in Deutsch steht da Meldungen und es bezog sich alles auf die v. 129. Wie unser Professor immer sagte:
Es gibt keine dummen Fragen, nur unqualifizierte Antworten.

stefan123t, 2024-07-12:
Also Hoymiles nennt die Struktur intern AlarmData und Alarm Code / ID und die Zeilen sind dann wie in der 0.8.129 Events aus dem EventLog.

Evtl. dauert es auch einfach eine Weile bis der WR der DTU auf die Frage nach dem AlarmData Frame / EventLog ordentlich geantwortet hat.
Dann sind es eben erst 0 und später 1 ?

Gubi2023, 2024-07-12:
nein, in der.128 wurde nicht hochgezählt, aber der Fehler wurde in der .129 berichtigt. Deshalb hab ich den Issue auch wieder geschlossen.