lumapu/ahoy #1705: ERR_DUPLICATE_INVERTER in 0.8.129 bei Speichern Invertereinstellungen

Loetnase, 2024-07-15:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.129

### Github Hash

3c0c8ee

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Neue 0.8.129,  

### Error description

In der neuen 0.8.129 kommt bei Speichern nach Anschauen der Invertereinstellungen ohne Änderungen die Fehlermeldung: ERR_DUPLICATE_INVERTER
Es ist auch kein zweiter Inverter eingetragen. 
@brekemeier Die Ursache ist bestimmt in der Ausmerzung von issue #1700 zu suchen
[image]

brekemeier, 2024-07-15:
Die Exzeption muss bereits bei der Eingabe der Inverter-Seriennummer erfolgen. Sie wird weiterhin (0.8.129) akzeptiert. Erst bei den Einstellungen (speichern) wird die Fehlermeldung erzeugt. 
[image]

Loetnase, 2024-07-15:
Ohne eingetragenen zweiten Inverter sollte aber diese Fehlermeldung gar nicht kommen.
Ich habe nur mein System über OTA auf die 0.8.129 hochgezogen, dann die Einstellungen des einen eingetragenen WR durchgeschaut, dann dort speichern gedrückt und die Fehlermeldung erhalten. 
Mit nur einem Inverter sollte die eigentlich nicht kommen.

brekemeier, 2024-07-15:
Ich aktualisiere das Build auch über OTA und kann zweimal die gleiche SN eingeben und erhalte keine Fehlermeldung (ESP32Wroom32).

AhoyDTU addiert auch die Werte, wie vorgesehen. Erst bei der Anpassung der PV-Module meldet er den Fehler.

Gut, ist ja nur ein Hinweis.  ERR_DUBLICATE_INVERTER sollte bereits bei der Eingabe der SN erfolgen. Sonst ist alles gut, ich bin bei

einem Test, für mehrere Inverter, darauf gestoßen und Fehler sollen wir ja melden.

knickohr, 2024-07-15:
#1700

knickohr, 2024-07-15:
Kann ich bestätigen. Egal was man an den Invertereinstellubgen macht, es kommt immer Duplicate. Vermutlich weil er den Inverter bereits kennt und beim Speichern eben nochmal die Seriennummer gespeichert werden will.

[image]

haraldhotzbehofsits, 2024-08-02:
Ich habe etwas overpanneling gemacht und wollte die Leistung des "Panels" ändern, und habe auch diesen Effekt beobachtet.

brekemeier, 2024-08-02:
Den Inverter löschen und neu einrichten! Schnellste Lösung!

Gubi2023, 2024-08-02:
oder die .127 flashen, mit der gehts!

knickohr, 2024-08-02:
Man kann natürlich auch das Settings-JSON editieren und neu importieren 🫣

Aber bitte, nur für die die auch genau wissen was sie da tun 😇

Loetnase, 2024-08-02:
Eventuell könnte es auch etwas bringen die nächste Version abzuwarten 😇

lumapu, 2024-08-04:
> Eventuell könnte es auch etwas bringen die nächste Version abzuwarten 😇

ja es geht endlich wieder weiter 😎 (hatte jetzt gute drei Wochen pausiert)

Loetnase, 2024-08-09:
Funktioniert wieder
Danke

closed 2024-08-09 by Loetnase