lumapu/ahoy #1707: [Enhancement] MQTT JSON Timestamp

MichiM2, 2024-07-21:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.127

### Github Hash

8cdf364

### Build & Flash Method

ESP Tools (flash)

### Setup

.

### Error description

1.) For saving and reporting reasons it would be useful, to include a timestamp in every MQTT JSON-packages:
like i.e. Tasmota does:    {"Time":"2024-07-21T15:24:01","U_AC": ...

knickohr, 2024-07-21:
Hier sind die atimestamps :

[image]

MichiM2, 2024-07-21:
Ja, schon. Aber die sind nicht *im* JSON. Ich archiviere .../ch0 bis .../chx für spätere Auswertungen. Da wäre es m.E. sinnvoll, wenn der Messzeitpunkt bei den zugehörigen Daten stehen würde und nicht in einem anderen topic.

geronet1, 2024-07-24:
Die ../chx Topics sind nicht retained. Somit ist der Messzeitpunkt der Zeitpunkt, wann du das Topic empfängst.

MichiM2, 2024-07-24:
Jein.
Ganz spitzfindig: das eine ist der Messzeitpunkt, das andere der Empfangszeitpunkt.
Meist stimmen die -zumindest in einem lokalen Netzwerk- natürlich überein. Das stimmt aber nicht mehr, wenn dazwischen ge-queued wird. Beispielsweise mit einem QOS=2 .
Meiner Erachtens gehört der Messzeitpunkt mit zu den Messdaten.

lumapu, 2024-08-04:
das ist was für unsere Liste #1199 (Wishlist)

lumapu, 2024-08-04:
habe den Timestamp eingebaut, finde es eine sinnvolle (einfache) Änderung.

So sieht das dann z.B. für CH1 aus:

```json
{
    "U_DC": 41.8,
    "I_DC": 8.49,
    "P_DC": 354.9,
    "YieldDay": 0,
    "YieldTotal": 123.456,
    "Irradiation": 88.725,
    "MaxPower": 354.9,
    "ts": 1722812839
}
```

closed 2024-08-16 by lumapu