lumapu/ahoy #1708: Frage zu API LimitSet mit CURL

dtuuser, 2024-07-21:
Hallo zusammen,
da ich in den letzten Tagen mehrfach das Problem hatte, dass mein Programm über CURL das Limit setzt, der WR das über Ahoy nicht umstellt. Dadurch regelt mein Programm ab diesem Zeitpunkt nicht richtig nach.

Laut "der" Suchmaschine, bekommt man bei einem CURL Aufruf über die Shell aus einem VB-Programm raus leider keine Rückmeldung, außer die ID des gestarteten Programmes.

Kenn jemand ein Work-Around?

dtuuser, 2024-07-21:
Habe eventuell eine Lösung gefunden.   
Über  /api/inverter/id/0   den Wert ""power_limit_read": 100," auslesen und mit in die Verarbeitung aufnehmen.
So quasi, bevor das nächste Limit gesetzt wird, da drüber auslesen und vergleichen, ob es gesetzt wurde.

stefan123t, 2024-07-21:
Ja richtig, das ActivePowerLimit braucht eijige Zeit bis es vom WR angenommen und bestätigt wurde. Sendet man bereits ein Neues Limit bevor der WR das Limit selbst bestätigt, dann kann er für längere Zeit nicht mehr antworten.
Das ist ein bekanntes Problem.

stefan123t, 2024-07-31:
@dtuuser schau Dir mal die Dokumentation zum [SystemConfigParam](https://github.com/lumapu/ahoy/wiki/Protocol#systemconfigpara--0x05) Command an, das erhält vom Inverter nur eine relativen % Wert zurück und ist entsprechend Timing sensitiv!

closed 2024-08-04 by dtuuser