lumapu/ahoy #1709: Daten via MQTT nach Stromausfall

Martin7025C, 2024-07-22:
Guten Morgen!

Hatte gestern mal wieder einen Stromausfall zuhause und danach ein kleines Problem mit dem AhoyDTU.

Und zwar, der AhoyDTU ist mit einer letzten gemessenen Leistung von 42W ausgefallen, sprich dies war der letzte Wert den der Loxberry MQTT empfangen hat und an Loxone sendete.

Nach dem Start war es schon dunkel und der Wechselrichter war offline, also keine neuen Werte mehr am MQTT, damit keine Änderung am Loxone sprich bis zum nächsten Tag 42W.

Jetzt wäre meine Frage ob es die Möglichkeit gibt beim Start des AhoyDTU einmal alle Werte vom Wechselrichter, auch wenn der offline ist und alles 0 ist an dem MQTT-Server zu senden. Dann würde ich am Loxone nicht die ganze nach 42W in der Visualisierung sehen. 

FWVersion ist die 0.8.83.

Vielleicht gibt es auch eine andere Lösung im Loxone die habe ich aber noch nicht gefunden.

Danke!

Gruß
Martin

knickohr, 2024-07-22:
Als erstes würde ich mal einen Update auf eine neuere Version machen. Wir sind schon fast 50 DEV-Versionen weiter, einige Probleme sind damit behoben.

Und in dem Fall das Dein Problem nicht gefixed ist, würde ich an deinem Loxone etwas Intelligenz einbauen. Ich kenne es leider nicht und auch nicht deren Möglichkeiten. Wir können aber nicht für jedes Tool die Eigenarten programmieren, das würde den Sinn von einer DTU widersprechen.

Schau mal unter den Invertersettings, vielleicht sind passende Optionen bereits vorhanden.

Martin7025C, 2024-07-22:
danke für den Tip, werde ich die Tage mal versuchen

geronet1, 2024-07-24:
Bei Nodered gibt es einen Trigger: Wenn 30 sek. kein Wert kommt, setze auf 0 oder NaN.

closed 2024-08-16 by lumapu