lumapu/ahoy #1712: [Bug] After using Console page the web interface is unresponsive

chmuc01, 2024-07-27:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

square dot

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.129

### Github Hash

3c0c8ee

### Build & Flash Method

ESP Tools (flash)

### Setup

n/a

### Error description

1) go to Ahoy-DTU main web page
2) select any page except the console to see that everything is working
3) select the console page (you can wait for some data to appear, but this is not necessary)
4) go to any other page: the interface is not responding any more, menu is reduced to the "AhoyDTU" and "Verlauf" menu with no data inside.

After about 10 minutes, the system reacts again (restarted itself).

I expected that leaving the console will not "crash" anything and that AhoyDTU still would work correctly without the need of a reboot.

lille2000, 2024-07-28:
See this issue #1619

lumapu, 2024-08-04:
have you tried to use the minimal version for ESP8266?

stefan123t, 2024-11-04:
@chmuc01 have you tried the v0.8.152 and used the minimal firmware image, i.e. without the graph ?
How many inverters have you configured in your AhoyDTU and what is the polling interval ?

chmuc01, 2024-11-05:
Sorry to say, but I didn't have the time to test something else. I immediately changed to an ESP32 as it seems that the 8266 just makes too much trouble. It would be good to just make an information that 8266 is not fully supported. 
The current prices of the 8266 and the ESP32 are so similar that it makes no sense to even think about saving 1-2€ but then just having trouble.
So from my side you can close the issue as I will not use a 8266 any more for AhoyDTU.