lumapu/ahoy #1713: nach Update kein 2.Login möglich [Bug]

jotroj, 2024-08-07:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.130

### Github Hash

GIT SHA: 24ceb3e :: 0.8.130

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nur Update von 0.8.129 auf 0.8.130

### Error description

Nur nach Power OFF/ON startet Normalbetrieb wieder für live und  Login

Gubi2023, 2024-08-09:
so, bei mir nun auch: Update .130 auf .131: DTU nicht mehr erreichbar! Anschluss an USB: Reboot in Dauerschleife!

Gubi2023, 2024-08-14:
mit .132 gehts

jotroj, 2024-08-19:
Update von 0.8.130 auf  0.8.141 : Login funktioniert