lumapu/ahoy #1714: bei ePaper Ahoy Logo fehlt beim Start bei 0.8.131

Loetnase, 2024-08-09:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.131

### Github Hash

55e45c7

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

bei der 0.8.131 mit **ePaper** fehlt beim Start bei das Ahoy Logo :disappointed_relieved: 

### Error description

bei der 0.8.131 mit **ePape**r fehlt beim Start bei das Ahoy Logo :disappointed_relieved:

Gubi2023, 2024-08-09:
kann es sein, dass die .131 nur funktioniert, wenn E-Paper angeschlossen ist?? Hab kein E-Paper, nach Update Dauerreboot!

Loetnase, 2024-08-09:
altes ESP8266 mit Nokia Display funzt auch mit 0.8.131.

Gubi2023, 2024-08-09:
wie gesagt, ich habe ESP32, kein Display, auch nicht aktiviert und mit .131 Dauerreboot!

Loetnase, 2024-08-09:
@Gubi2023 Probiere es mal mit den Einstellungenfürs ePaper, natürlich ohne ePaper dran
Vielleicht funzt es dann.
[image]

Gubi2023, 2024-08-09:
nö, ich warte auf die nächste Version, das wäre eh nur ein workaround, von dem ich nicht weiss, ob er funktioniert.

Gubi2023, 2024-08-09:
danke, update mit .132 funzt! =))

Loetnase, 2024-08-09:
Habe auch gerade die 0.8.132 eingespielt, 
das Logo kommt auch beim Start :thumbsup: 
Würde auch sagen funzt :innocent: 

@lumapu Dankeschön

Loetnase, 2024-08-13:
Das funzt 😇
und ich mache hiermit das Issue zu.
@lumapu Dankeschön

closed 2024-08-13 by Loetnase