lumapu/ahoy #1715: [Bug] MQTT - Total der ahoyDTU wird nicht im Discovery Config (homeassistant) mitgesendet

evilmumi, 2024-08-09:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.130

### Github Hash

24ceb3e

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

[image]

### Error description

Nach einem Restart der dtu / homeassistant habe / hatte ich immer mal Probleme mit unavailable von den Sensoren in HA.
Es half immer ein manuelles drücken des SEND Discovery Config (homeassistant).
Dies funktioniert nun auch für alle beiden Inverter.
Diese tauchen unter homeassistant/sensor/ auf.
Leider nicht (mehr) das ahoydtu ansich - ich habe kürzlich von 0.8.85 auf .130 geupdated. Ich weiß nicht, ob es damit zusammen hängen könnte. Ich habe erstmal den issue eröffnet bevor ich wieder downgrade.

evilmumi, 2024-08-09:
Update: ein Downgrade auf 0.8.85 hilft - das discovery sendet homeassistant/sensor/ahoydtu32_TOTAL.

evilmumi, 2024-09-17:
hat wer einen Tipp hierzu? das downgraden nur für den Publish ist nervig. Ich bin aber nicht so der mqtt experte - aber kann ich nicht die Sachen selber publishen? jemand eine Idee?

martin-st-81, 2024-09-21:
Ist per MQTT in HomeAssistant verfügbar, aufgrund des Bugs muss es manuell in der mqtt.yaml hinzugefügt werden.
Ich habe es wie u.a. eingebunden:

sensor:
    - name: Aktuelle Leistung Solar
      state_topic: "inverter/total/P_AC"
      qos: 2
      unit_of_measurement: "W"
      unique_id: "AktuelleLeistungSolar"
      state_class: measurement
      device_class: energy
      icon: mdi:solar-power

evilmumi, 2024-09-23:
welchen Bug meinst du?

und wie bekomme ich das hin, dass es regelmäßig gesendet wird? Manchmal ist es nach HA restart /reboot/mqtt server reboot weg!? ich spiele mal 145 ein

lumapu, 2024-09-23:
Ich bin kein homeassistant-Nutzer, daher kann ich die Frage nicht beantworten. Muss man das wirklich regelmäßig senden?

martin-st-81, 2024-09-24:
die Übermittlung mittels MQTT nach HomeAssistant erfolgte korrekt (ersichtlich in MQTT Explorer im HomeAssistant). Nur wurde es nicht autom. eingebunden. Dies wurde mit Version 145 jetzt behoben.

evilmumi, 2024-09-24:
wie gesagt, ich beobachte nun mit 145 und melde mich, wenn es nicht mehr geht, was ich getan habe. danke soweit

stefan123t, 2024-11-04:
@evilmumi hast Du das Problem mit der aktuellen v0.8.152 noch oder können wir das issue als gelöst schließen ?

evilmumi, 2024-11-04:
Ich habe gerade die .145 drauf
Wollte gerade auf 152 updaten. Da sagt er failed. Weiß nicht warum. Nach einem reboot macht er aber auch hier nicht nochmal eine home Assistent Ankündigung somit sind die Werte alle leer.

evilmumi, 2024-11-04:
Ich denke, es wäre super, wenn bei einem neustart von ahoy und oder HA dies neu gesendet wird. Bin aber nicht so tief in dem Thema um das bewerten zu können. Naja die, die kein MQTT + HA haben könnten es ja per checkbox abwählen.
Ich überlege auch gerade, ob es eine Möglichkeit gibt, dass ich nach einem HA Neustart die routine bei ahoy anstarten könnte - das wäre ja auch nen (Super.)Weg (für mich)

AndiUCG, 2025-01-22:
Hi, ich verliere jede Nacht die Verbindung im HA. Erst nach einem manuellen Klick auf Discovery taucht er weider auf. Ist das zufällig ein ähnliches Phänomen wie hier beschrieben? Wenn ja gibt es dafür schon eine Lösung? Ich nutze Version 140

Gubi2023, 2025-01-22:
versuch doch mal die .152
Lt. lumapu wurde mit .145 das gefixt..

evilmumi, 2025-01-22:
mit  .151 ist es nicht viel besser.
auf .152 kam ich noch nicht da es online nicht geht und offline noch keine Zeit war