lumapu/ahoy #1717: [Bug] v0.8.134 Deutsche Version teilweise nicht übersetzt (System)

Gubi2023, 2024-08-11:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.134

### Github Hash

cd16b37

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Update von v127 auf v134

### Error description

Deutsche Version teilweise nicht übersetzt auf überarbeiteten System Unterseite.

lumapu, 2024-08-11:
😂 ich hab schon drauf gewartet. Wird früher oder später behoben

Gubi2023, 2024-08-11:
nix für ungut, wollt es nur mitteilen.

Gubi2023, 2024-08-11:
na, das ging ja fix :)) aber ein paar Kleinigkeiten hast du übersehen:  
[image]
[image]
[image]

lumapu, 2024-08-11:
bei Windows geht das mit WIN Taste und Punkt

Gubi2023, 2024-08-16:
Hallo Lukas, zu früh zugemacht, die letzten 3 Fehler sind noch drin...

lumapu, 2024-08-16:
es ist immer zu früh 😅, irgendwann muss man diesen Schritt gehen (nach einem halben Jahr mal wieder)

Gubi2023, 2024-08-17:
Ok

closed 2024-08-17 by Gubi2023