lumapu/ahoy #1718: Schönheitsfehler(?)  Warnmeldung beim update

Gubi2023, 2024-08-11:
Schönheitsfehler(?) beim update derselben Version von deutsch auf englisch oder umgekehrt erscheint Warnmeldung
"Die ausgewählte Firmware passt u.U. nicht zum Chipsatz!"  das irritiert doch manchen user...

lumapu, 2024-08-13:
normalerweise sollte die Sprache gleich bleiben denke ich

Gubi2023, 2024-08-14:
habe gerade von 240813_ahoy_0.8.137_8f12bfe_esp8266.bin  auf esp8266-all (Module: MDH-en) upgedatet (also gleiche Sprache)
Da kommt die Meldung  überflüssigerweise ebenso

knickohr, 2024-08-15:
Das ist hiervon abhängig. Wenn es da eine Änderung gibt, kommt diese Warnmeldung. It‘s not a bug, it‘s a feature. Um Deine grauen Gehinnzellen noch,so anregen nachzudenken ob Du das auch wirklich willst 😅

[image]

Gubi2023, 2024-08-15:
mich störts ja weniger, aber mancher Anwender wird dadurch verunsichert, wie ich schon einigen Kommentaren entnommen habe....

knickohr, 2024-08-15:
Stimmt auch wieder. Vielleicht wäre eine kurze Erklärung bei der Meldung zielführend.

Gubi2023, 2024-08-23:
ich würde mir wünschen, dass ihr die Auslösung der Meldung doch nochmal überdenkt, ob nicht andere Kriterien ausreichen. Dass die Meldung bei gleicher Konfig der DTU kommt, ist schon störend/irritierend....

stefan123t, 2024-11-04:
Wie wäre es die Meldung anzupassen: 
"Die ausgewählte Firmware passt u.U. nicht zur bestehenden Firmware (bisherige Firmware %s; neue Firmware %s)"

Bei mir ist das Ergebnis Environment: `opendtufusion (build options: MDH-en)`.
Eventuell kann man ja die Build Options nicht als Teil der Umgebung bzw. des Build Targets betrachten sondern separat mitspeichern oder nur den Teil vor der Klammer mit den build options vergleichen ?

https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/src/web/html/update.html#L52-L65