lumapu/ahoy #1719: [Bug] Interrupt bin bleibt auf false nach update auf 8.127+ / keine Verbindung zum Inverter

Falerion-git, 2024-08-11:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

opendtu fusion board (already assembled)
hoymiles HMS-2000 (black version)

### Debug Serial Log output

```shell
CMT TX failed, code: 2
```

### Error description

Ich habe ein  Problem, dass der Interrupt als "false" angezeigt wird (opendtufusion board).

Mir ist aufgefallen, dass mit den dev Versionen 8.127, 8.130 oder 8.134 (andere nicht ausprobiert) der Interrupt Pin als "false" angezeigt wird, ähnlich zu einigen anderen issues hier. Wenn ich auf die Version 8.83 zurück kehre, dann bleibt der Interrupt Pin zunächst auf "false".
Erst wenn ich in den Einstellungen des Inverters unter "Radio" die "Frequency" auf 866.00 Mhz setze bekomme ich den Inverter direkt wieder angezeigt. Nach dem downgrade von den dev Versionen ist die zunächst auf 860.00 MHz eingestellt.
Vor dem Upgrade hatte ich auch die Frequenz von 866.00 MHz genutzt.

Der Fehler code war wie im issue oben: CMT TX failed, code: 2
Nach dem Frequenzwechsel ist dieser Fehler auch verschwunden.

Ich bin ursprünglich auf eine der dev Version gewechselt, weil ich jetzt einen zweiten Inverter einbinden wollte, der eine Seriennummer mit Buchstaben hat. Leider schient es die Option, die Frequenz einzustellen nicht mehr zu geben (zu mindestens in den ausprobierten dev Versionen (siehe oben)). Der ursprüngliche Inverter wird mir in der Übersicht als "enabled" angezeigt, aber er liefert keine Daten. Wenn ich die "Active Power Control" aufrufe und dann einen "Restart" des Inverters anstoße, bekommen ich eine "Error: authentification failed" Fehler. Vermutlich aufgrund des nicht funktionierenden Interrupts.

Wird die Frequenz jetzt eigentlich automatisch eingestellt?
An den Pinouts kann es doch eigentlich nicht liegen, oder? Ich hatte auch versucht, die nach der Lösung in einem anderen issue zu tauschen (in den Einstellungen), aber ohne Erfolg.
Kann ich in den dev-Versionen irgendwie die Frequenz manuell einstellen?

unabhängig vom obigen Problem:
Bei den dev Versionen ist mir aufgefallen, dass wenn ich einen zweiten Inverter mit Buchstaben Seriennummer hinzufüge, ich den nicht aktivieren kann. (Erster Inverter ohne Buchstaben in der Seriennummer). Ein setzen der "Enabled" Box wird beim speichern ignoriert.
Zudem wird immer nur ein "Input" (also die Panel Eingänge) als konfigurierbar angezeigt. Weitere "Input" Felder lassen sich nicht hinzufügen, bzw. sind nicht vorhanden.

Gubi2023, 2024-08-11:
hast du mal versucht, die DTU über https://fw.ahoydtu.de/tools/webflasher/  neu zu flashen? da wird beim opendtufusion die pinconfig automatisch eigespielt...

Dirk-Ahoy, 2024-08-11:
Wenn du die Frequenz im laufenden Betrieb ändern willst musst du die DTU mindestens 15min abklemmen - erst danach hört der HMS wieder auf einer anderen (neuen) Frequenz.

In den letzten von dir genannten Versionen ist nichts an entsprechender Stelle geändert worden.  Die letzen Änderungen am Funkprotokoll sind schon lange her. 

Das mit dem CMT TX failed hab ich früh morgens oder nach einer Freq. Änderung auch hier und da mal aber geht trotzdem.

Villeicht wirklich mal einen kompletten reset machen und alles neu einstellen.

Falerion-git, 2024-08-11:
> hast du mal versucht, die DTU über https://fw.ahoydtu.de/tools/webflasher/ neu zu flashen? da wird beim opendtufusion die pinconfig automatisch eigespielt...

Ja, das habe ich probiert, mit und ohne erase. Scheint keinen Unterschied zu machen.

> Frequenz kannst du beim HMS/HMT einstellen
Ja, so mache ich das bei der 0.8.83 version. 
Ich meine, dass ich das Menü habe ich aber nicht angeboten bekommen bei den 0.8.127+ Versionen, dort konnte ich nur eine Range 860-870 MHz einstellen. Ändert sich das Menü im laufenden Betrieb? Ansonsten muss ich das nochmal ausprobieren, evtl. habe ich das nur übersehen. Werde ich hier nochmal posten.

Wie läuft die Frequenzfindung denn typischerweise von Seiten des Hoymiles ab? 
(ich bin nicht mehr sicher, warum ich damals 866.00 MHz gewählt habe, bzw. warum das vor eingestellt war)

Dirk-Ahoy, 2024-08-11:
Wie geschrieben wenn du die DTU mindestens 15min abschaltest kannnst du mit jeder Freq. probieren die zur Auwahl steht - ist ja durchs Land beschränkt.  Oder halt Nachts umstellen für den nächsten morgen.

Im laufenden Betrieb umstellen geht nicht da der WR dann immer nur die vorher eingestellte abhört.

lumapu, 2024-08-11:
> Wie läuft die Frequenzfindung denn typischerweise von Seiten des Hoymiles ab? (ich bin nicht mehr sicher, warum ich damals 866.00 MHz gewählt habe, bzw. warum das vor eingestellt war)

Ich glaube mich zu erinnern, dass die auf 868 MHz umstellen. Der Wechselrichter startet mit 863MHz.

Dirk-Ahoy, 2024-08-11:
Ich meine die nehmen 2 Frequenzen beim Start ist auf jedenfall 868 dabei.  Müsst ich morgen früh mal gucken wenn ich früh genug auf bin :)

lumapu, 2024-08-11:
ja stimmt, sie funken auf 863MHz, dass der Wechselrichter doch bitte auf 868MHz wechseln soll. Auf 868MHz fragen sie dann die Daten ab, deher die beiden Frequenzen.

Falerion-git, 2024-08-12:
Ich habe es heute nochmal getestet. 
Panels abgeklemmt, AhoyDTU neu geflasht und die Settings wieder eingelesen. Dann die Panels wieder angehängt. 
Dann die Frequenz manuell auf 866.00 MHz (bei mir nicht 868.00 MHz) stellen und es funktioniert auf Anhieb.

closed 2024-08-12 by Falerion-git