lumapu/ahoy #1720: [Bug] ESP32 nach Vergabe von fester IP Adresse nicht mehr erreichbar

RealNBB, 2024-08-12:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.135

### Github Hash

83b386d

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Vergabe feste IP Adresse 

### Error description

Hallo,

nach einem Update von 0.8.129 auf 0.8.130 hatte ich den Bootloop und musste daher nochmal komplett neu flashen. Da ich paar Tage unterwegs war und nicht dazu kam, erschien mittlerweile die 0.8.135 und daher wollte ich direkt auf diese Version gehen.

Den Wizzard ist noch durchgelaufen – mit Eingabe des Wlan-Schlüssels – aber als ich allerdings im zweiten Schritt eine feste IP vergeben wollte, war die DTU nach dem obligatorischen Reboot nicht mehr erreichbar. Das Ganze habe ich 2x probiert – immer mit dem gleichen Ergebnis.

Jetzt läuft Version 0.8.132 auf dem ESP32 ohne Probleme, da ich vermute, dass es etwas mit den Ethernet-Settings zu tun hat, welche ab Version 0.8.133 Einzug gehalten haben.

Dirk-Ahoy, 2024-08-12:
Bitte maximal 133 nehmen in der 134 und 135 ist ein Bug der die Settings kaputt macht  beim speichern. Dann ist das Passwort vom Wlan und AP Mode weg.

RealNBB, 2024-08-12:
Alles klar, Danke für die Info!

closed 2024-08-16 by lumapu