lumapu/ahoy #1724: [Bug]HMS-1600-4T causes reboot on latest DEV versions after ~30sek

stefanbode, 2024-08-15:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.132

### Github Hash

c64f028

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT setup. Is already working with a HM 1600. 
Try to add a second inverter with following serial: 1164A00924ba

### Debug Serial Log output

```shell
14:43:07.027 I: [NTP]: 2024-08-15 14:43:07 UTC
14:43:07.075 I: MQTT connected
```

### Error description

device reboots every 30 seconds. In webserial the clock in the bottom doe not count more than 40 sec.

If I disable the new  INVERTER1 everything is working again. If I switch INVERTER0 and INVERTER1 it also does not work. If I remove INVERTER0 with the new serial it also does not work.

The reboot is the same no matter if the HMT is physical up and running or OFF and diconnected.

lumapu, 2024-08-15:
can't be ESP8266 if you have a HMT inverter involved.
For ESP32 there is a button in the section `/system` to download a coredump. This file would be of interest to check where the ESP crashes.

You can also write German if you like

stefanbode, 2024-08-15:
Ok great. I have enough ESP32. Will replace it tomorrow and see if this makes a difference.

stefanbode, 2024-08-16:
HMS and HMT do require ESP32 and a different communication module.

closed 2024-08-16 by stefanbode