lumapu/ahoy #1725: [Bug]

Gubi2023, 2024-08-15:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.138

### Github Hash

01285f0 

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update von .137 auf 1.38

### Error description

nach update zuerst Indexseite ohne Inhalte, nach Spannung aus/ein eine Zeitlang reguläre Anzeige, dann wieder leere Anzeige und nach Refresh Fehlermeldung 
[image]

lumapu, 2024-08-16:
ist das die minimal version?

Gubi2023, 2024-08-16:
Gerätename | AHOY-DTU
-- | --
0 days, 00:03:49
Software Watchdog
esp8266-all-de (Module: MDH-de)
0.8.138 - 01285f0
CPU: 80MHz, undefined Core(s)

DTU startet sich jetzt alle paar Minuten neu!

Gubi2023, 2024-08-26:
ich habe jetzt die neueste -min geladen, leider immer noch DTU-Neustarts...
[image]

stefan123t, 2024-11-04:
@Gubi2023 ist das Problem mit der aktuellen v.08.152 immer noch aufgetreten ?

Der o.a. Content-Encoding Fehler tritt vermutlich auf, wenn die komprimierten CSS oder JavaScript Daten der AhoyDTU Live Webseite, etc. nicht vom ESP geladen werden können. Hierbei wird m.W. die Dateien bereits komprimiert im Filesystem des ESP gespeichert und dann beim senden lediglich der `Content-Encoding: gzip` Header gesetzt. 

Du kannst das ggf. mit den **Developer Tools** (F12 Taste) > **Network** und einem einem **_v_ Export HAR (sanitized)** des Vorgangs genauer und besser belegen. Da sehen wir dann die einzelnen Requests und Responses von Deiner AhoyDTU.

[image]

Gubi2023, 2024-11-04:
ESP8266 mit "240816_ahoy_0.8.140_f1f4481_esp8266-de.bin"  läuft seit 20 Tagen störungsfrei.  Die all-Version mit MqTT stürzte regelmässig ab. Ich bin nur Anwender und kein Entwickler, deshalb kann ich deine oben genannte Fehlersuche nicht durchführen (mir zu kompliziert).