lumapu/ahoy #1726: [Bug]

RealNBB, 2024-08-16:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.130

### Github Hash

24ceb3e

### Build & Flash Method

was already installed

### Setup

Nicht geändert. Meine Standardeinstellung wie in den Versionen zuvor.

### Error description

Hallo,

ich nutze das Skript "HoymilesZeroExport", um eine Nulleinspeisung zu realisieren. Das Skript regelt ja bekanntlich die Leistung der einzelnen Inverter; bei mir in Summe drei Stück.
Ab Version 0.8.130 steigt bei mir nach einiger Zeit das WebIf aus bzw. gleich der ganze Inverter, heißt, er verharrt auf einer x-beliebigen eingestellten Leistung und ist dann nicht mehr erreichbar/regelbar.

Version 0.8.129 lief bis dato ohne Probleme mit den identischen Settings wochenlang absolut stabil durch.

lumapu, 2024-08-16:
kannst du bitte mal diese Firmware testen: https://github.com/lumapu/ahoy/issues/1685#issuecomment-2282879153

Diese hat einen anderen Webserver, den ich gerne in Zukunft einsetzen würde.

Ollipop030, 2024-08-17:
> Ab Version 0.8.130 steigt bei mir nach einiger Zeit das WebIf aus bzw. gleich der ganze Inverter, heißt, er verharrt auf einer x-beliebigen eingestellten Leistung und ist dann nicht mehr erreichbar/regelbar.

Exact das Gleiche passiert bei mir auch. ESP32 mit 2 Invertern und HoymilesZeroExport. Alles >130 steigt nach ein paar Stunden aus, die Versionen darunter laufen tagelang durch. Neue Version teste ich auch mal.

SnakeGER, 2024-08-17:
Habe ein ähnliches Problem.
Bei mir werden 3 Hoymiles über MQTT geregelt (Akkustand Speicher ~ MQTT Raspberry ~ AHOY).
9 von 10 Limit-Befehle werden nicht auf allen 3 WR umgesetzt. Die Befehle kommen zwar laut Konsole auf dem AHOY an, aber bei mindestens einem WR steht dann sowas wie "CMT tx fail".
Bis zur version 0.8.129 lief das aber wunderbar und es gab auch keine technischen Veränderungen.
Habe das gerade nochmal mit der 0.8.141 getestet. Problem bleibt.
Ich bleibe jetzt erstmal bei der 0.8.129 bis das Problem behoben ist.
LG

lumapu, 2024-08-17:
@SnakeGER naja, wenn ich ins changelog schaue wurde schon was geändert, wahrscheinlich meintest du techn. Änderungen auf deiner Seite. 

Auszug aus dem Changelog, was sich mit deiner Beobachtung decken könnte:

## 0.8.130 - 2024-08-04
* merge PR: Power limit command accelerated #1704

[Changelog](https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md#08130---2024-08-04)

SnakeGER, 2024-08-17:
@lumapu natürlich meine ich auf meiner seite. Aber das mit dem "Power limit command accelerated" könnte natürlich zu dem Fehler führen, aber davon hab ich zu wenig Ahnung 😆
Jedenfalls muss etwas geändert worden sein, denn bis zur 129 gabs den Fehler nicht

reserve85, 2024-08-18:
Ich hatte das bis gestern nicht feststellen können. 
Habe gestern dann aber ein mal den Fall gehabt, dass ein Limit set auf 100% als accepted gemeldet wurde aber der inverter weiter bei dem letzten wert verblieb. Keine Ahnung ob das damit zusammenhängt? Ansonsten gabs bei mir keine Abstürze o.Ä.. aktuell bin ich auf der 140.

RealNBB, 2024-08-18:
> kannst du bitte mal diese Firmware testen: [#1685 (comment)](https://github.com/lumapu/ahoy/issues/1685#issuecomment-2282879153)
> 
> Diese hat einen anderen Webserver, den ich gerne in Zukunft einsetzen würde.

Hi....also sowohl die von dir verlinkte Version als auch die neue 141 funktioniert nun ohne Aussetzer! Von meiner Seite also alles wieder top! :)

Dank Dir und einen schönen Sonntag!

SnakeGER, 2024-08-18:
Bei mir leider immernoch das Problem mit den nicht umschaltenden WR.
Aber die 0.8.129 läuft gut und ich warte erstmal ob sich in den kommenden Updates etwas ändert.
Sonst mache ich mal einen eigenen Thread auf. Hab mich ja hier nur eingemischt ;)

Trotzdem nochmal großes Lob an die Entwickler, dass ihr sowas tolles auf die Beine gestellt habt. Hut ab

Ollipop030, 2024-08-19:
Bei mir läuft die 141 leider auch nicht. Gestern gegen 16:00 neu geflasht, heute morgen war die DTU noch online, kurz nach Acht kommt in HoymilesZeroExport nur `Aug 19 08:07:05 Pi-Hole python3[24517]: 2024-08-19 08:07:05 WARNING  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='192.168.0.19', port=80): Read timed out. (read timeout=10)")': /api/inverter/id/1`
WebInterface ist dann auch weg.

ArnoldSchrader, 2024-08-19:
Ich habe ein ähnliches Problem; ab Version 0.8.130 läuft die Ansteuerung per API nicht mehr zuverlässig. Ich vermute, das Problem tritt nur bei mehr als einem konfigurierten Inverter auf, da ich es nur dann beobachtet habe, wenn mehr als ein Inverter per API geändert werden sollen: Es wird zwar ein ok jeder der Änderungen zurückgegeben - aber diese passiert beim 2. dann scheinbar nicht.
Ignoriert die #1704 ggf. bei zwei gleichzeitig vorhandenen Änderungen die 2.?

ArnoldSchrader, 2024-08-20:
@BlackcatSandy Okay, war auch mehr geraten als gewusst ;-) Dann wundert mich aber, dass hier nicht deutlich mehr betroffen sind?!

Ollipop030, 2024-08-20:
> Dann wundert mich aber, dass hier nicht deutlich mehr betroffen sind?!

Ich schätze mal, nicht jeder hat immer die aktuelle Dev Version installiert, und von denen hat ein noch kleinerer Teil eine Nulleinspeisung laufen. Und es ist ja noch Urlaubszeit, viele chillen vielleicht noch in der Sonne. ;-)

dtuuser, 2024-08-20:
Ich fahre die Nulleinspeisung über ein eigenes Programm mit CURL Aufruf.
Aktuell noch mit Version .113, da alle neueren DEVs ab und zu mal ein Limitsetzen vergessen bzw. vom WR nicht angenommen werden.
Arbeite gerade an dem Auslesen des aktuellen Limits und wenn das eine Differnz zum vorher geschickten hat, dann wird das Limit nochmals geschickt.  Alle Tools laufen unter Windows

dtuuser, 2024-08-20:
Problem der auslesenden Rückmeldung.....dauert bis zu 2 Minuten.
Aktuell lese ich alle 10 Sekunden mit Tasmota den Stromzähler aus und regel dann alle 60 Selunden den WR. 
So habe ich eine Einspeisung von knapp 5% vom Ertrag hinbekommen.

Ollipop030, 2024-08-21:
Was mich aber wundert: Bei euch scheint die 141 ja zu laufen, bei mir stürzt die Version alle paar Stunden ab. Übrigens auch die Versionen davor. Das Problem fängt bei mir ab der 130 an, alles davor läuft wunderbar über Wochen.

lumapu, 2024-08-29:
⚠ Achtung, Version 0.8.142 funktioniert nicht, siehe

* #1743

ArnoldSchrader, 2024-09-01:
So, ich muss noch berichten - aber leider nichts positives: Habe die 0.8.143 ausprobiert und muss leider sagen, dass ich keinen Unterschied bei API-Stabilität zur >=0.8.130 feststellen kann. Lediglich bei den HTTP-Requests ist es nun "empfindlicher" (setSync muss false und setNoDelay true sein, sonst werden POSTs völlig ignoriert). Aber auch mit der 0.8.143 muss etwa 40-50% der API-Posts min 1x wiederholt werden (also tlws. bis zu 10x/Min). Auf 0.8.129 war das vllt 1x in der Stunde erforderlich.

Hinweis: ich habe in den Einstellungen bei Wechselrichter 1s als Intervall. Ich werde wohl erstmal bei 0.8.129 bleiben müssen..