lumapu/ahoy #1728: History menu disappeared

jschwed, 2024-08-17:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.140

### Github Hash

ahoy_v0.8.140

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Setup not changed during update from 0.8.83 to 0.8.140

### Error description

Hi,
I was running 0.8.83 and updated to 0.8.140.
After the update "History" in the menu is gone.  
[image]

Before it was listed right after "Live".
Under Settings/Protection I can still see the option to protect the History page. 
Tried to protect the page, reboot --> no change. Changed back to unprotected, reboot --> no change.

Do I need to change the config somewhere with this release?

thx

Dirk-Ahoy, 2024-08-17:
https://fw.ahoydtu.de/fw/download/release%2Fahoy_v0.8.140%2FESP8266%2F240816_ahoy_0.8.140_f1f4481_esp8266-all.bin

Please install this Version  :)

jschwed, 2024-08-18:
Hi @Dirk-Ahoy , thanks a lot for the quick response! With the "all.bin" it works. 
I used the 240816_ahoy_0.8.140_f1f4481_esp8266.bin file in the first place.

@lumapu: the new graphs look great - the new "Total Power Today" is very useful!

closed 2024-08-18 by jschwed