lumapu/ahoy #1730: [Bug] 0.8.140 Config export leads to misleading message "reboot AhoyDTU to apply all your configuration changes"

fsck-block, 2024-08-19:
Before exporting:
[image]

After export of configuration
[image]

Seems to be a side effect of
```
void onDwnldSetup(AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response;

    // save settings to have latest firmware changes in export
    mApp->saveSettings(false);`
```

Gubi2023, 2024-08-19:
siehe  #1721

fsck-block, 2024-08-19:
> Feature, no bug!
Könntest du mir erklären wo das das Feature ist?

Ich speichere meine Änderungen und mache einen Reset.
Nach dem Neustart exportiere ich nun die funktionierenen Einstellungen
Ich werde aufgefordert einen Reset durchzuführen um meine Änderungen zu aktivieren.
Nach dem Neustart exportiere ich nun die funktionierenen Einstellungen
Ich werde aufgefordert einen Reset durchzuführen um meine Änderungen zu aktivieren.
...

Gubi2023, 2024-08-19:
wurde hier #1454  schon vor 6 Monaten ausfühlich diskutiert!

fsck-block, 2024-08-20:
> ja ihr hab es gefunden 😊. Werde ich erst mal nicht ändern, wollte an den Einstellungen sowie einiges umbauen, sodasss nur noch selten ein Neustart erforderlich ist, bis dahin bleibt es so wie es ist.

Wenn ich @lumapu richtig verstanden habe, ist das kein Feature und er wollte #1454 offen lassen bis die Einstellungen umgebaut hat. 
Na dann lassen wir doch diesen Issue solange offen  :wink: