lumapu/ahoy #1731: V  0.8.141 : nach Login keine Menü-Auswahl verfügbar

jotroj, 2024-08-19:
### Platform

ESP8266

### Assembly

I did the assembly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.141

### Github Hash

GIT SHA: 85f8540 :: 0.8.141

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

keine

### Error description

1.)Nach 2.Login nur mehr 
AhoyDTU
Verlauf
REST API
Dokumentation
Website
Über
          sichtbar.
2.) Dies erfolgte nach erfolgreichem Login nach Updated auf 0.8.141. vieleicht 1 Minute davor waren alle Einträge "Daten, Verlauf, Einstellungen.." sichtbar und wurden auf obiges zurückgesetzt.
3.)Nach Hardwarereset lässt sich das wiederholen.

Dirk-Ahoy, 2024-08-19:
dann klemmt was bei dir im Wlan - passiert hier und da schon mal.  Einfach abwarten oder Browser neu starten wenn  es nach 10sec nicht wieder kommt.

lumapu, 2024-08-28:
ist der Fehler noch da?