lumapu/ahoy #1732: Ahoy issues with CMT2300A on latest relase 0.8.140 no data received

Alf-Ele, 2024-08-19:
### Platform

ESP32

### Assembly

I did the assembly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.140

### Github Hash

240816_ahoy_0.8.140_f1f4481_esp32-wroom32-de

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Ahoy DTU with CMT2300A 2x HMT2250

### Error description

With Version 0.8.83 ahoy worked fine, after upgrade to 0.8.140 no data received from CMT or it takes up to several minutes.
RF Interface seems to be instable.
CMT2300A is recognized and connected Int-Pin is working.
I could tag the issue down to 0.8.102 working fine with 0.8.103 onwards it takes very long until data is received.
CMT changes are in log of 103 and 106.
My Ahoy system is running in parallel to an OpenDTU
Both are accessing the same set of 2 HMT2250 on the same frequency.
With release 08.083 until 08.102 it works fine and stable.
Some changes introduced with 103 and later seems to have a negative input to CMT2300A RF interface.

[image]

Gubi2023, 2024-08-19:
siehe  https://github.com/lumapu/ahoy/issues/1384#issuecomment-1905799203  😊

Alf-Ele, 2024-08-19:
Mit 0.8.84 lief es ja stabil 2 Monate. d.h. ich muss bei der Version (bzw. 0.8.102 seit heute )bleiben?

DanielR92, 2024-08-19:
Es ist nicht ratsam zwei DTUs parallel im Betrieb zu haben (punkt). Es kann kurzzeitig gehen, aber es werden immer Probleme da sein.

Alf-Ele, 2024-08-19:
ok, Danke für die Antwort. Ich suche mir eine Alternative.
Der Punkt kann dann geschlossen werden

Alf-Ele, 2024-08-19:
Habe jetzt das "andere DTU" deaktiviert und das Verhalten ist identisch:
SW 0.8.140:
16:53:39.110 -----
16:53:39.112 W: CMT TX failed, code: 2
16:53:39.151 -----
16:53:39.152 W: CMT TX failed, code: 2
16:53:39.188 -----
16:53:39.189 W: CMT TX failed, code: 2
16:53:39.299 I: (#1) RX  57ms | 14 -72dBm | d6 02
16:53:39.437 I: (#1) RX  71ms | 27 -72dBm | 95 01
16:53:39.622 I: (#1) RX  68ms | 27 -72dBm | 95 02
16:53:39.755 I: (#1) RX  68ms | 27 -72dBm | 95 03
16:53:39.757 I: (#1) timeout, no attempts left
16:53:39.758 -----
16:53:39.759 I: com loop duration: 759ms
16:53:39.760 -----
16:54:07.072 -----
16:54:07.074 W: CMT TX failed, code: 2
16:54:07.146 -----
16:54:07.147 W: CMT TX failed, code: 2
16:54:07.450 I: (#1) timeout, no attempts left
16:54:07.451 -----
16:54:07.546 I: (#1) RX  82ms | 27 -72dBm | 95 01
16:54:07.700 I: (#1) RX  87ms | 27 -72dBm | 95 04
16:54:07.809 I: (#1) timeout, no attempts left
16:54:07.810 -----
16:54:07.811 I: com loop duration: 811ms
16:54:07.812 -----

SW 0.8.102
16:58:05.224 I: (#0) RX  84ms | 27 -72dBm | 95 81
16:58:05.228 -----
16:58:05.523 I: (#0) RX  80ms | 27 -72dBm | 95 01
16:58:05.524 I: (#0) RX 130ms | 27 -72dBm | 95 02
16:58:05.526 I: (#0) RX 180ms | 27 -72dBm | 95 03
16:58:05.527 I: (#0) RX 230ms | 27 -72dBm | 95 04
16:58:05.528 I: (#0) RX 280ms | 27 -72dBm | 95 05
16:58:05.725 I: (#0) RX  70ms | 15 -72dBm | 95 87
16:58:05.959 I: (#0) RX  68ms | 27 -72dBm | 95 06
16:58:05.962 -----
16:58:06.309 I: (#1) RX  81ms | 27 -72dBm | 95 01
16:58:06.310 I: (#1) RX 130ms | 27 -72dBm | 95 02
16:58:06.311 I: (#1) RX 180ms | 27 -72dBm | 95 03
16:58:06.312 I: (#1) RX 230ms | 27 -72dBm | 95 04
16:58:06.313 I: (#1) RX 280ms | 27 -72dBm | 95 05
16:58:06.325 I: (#1) RX 330ms | 27 -72dBm | 95 06
16:58:06.757 I: (#1) RX  59ms | 15 -72dBm | 95 87
16:58:06.761 -----
16:58:06.762 I: com loop duration: 1762ms
16:58:06.762 -----

Und ja der Empfang ist schwach, das Gerät steht in einem Gebäude, die Wechselrichter sind 20m entfernt.
Mal sehen was andere CMT2300A Nutzer beobachten ;-)

DanielR92, 2024-08-20:
Ich nutze selbst ein CMT. DTU zu Hoymiles trennen ca. 10meter Luftlinie, dazwischen 30cm Mauerwerk, Trapezdach. 
Bei mir funkt es auf der Freq.: 863.25 MHz (aktuell 20dBm (zum testen).

[image]

Vielleicht hilft dir die Info bei etwas?

Florian89163, 2024-08-21:
Gleiches Problem mit 0.8.140 auf ESP32-S3 bei mir

0.8.140 - f1f4481
---------------
19:30:18.001 I: (#0) Radio infos: t: 96, s: 0, f: 0, n: 96 | p: 20
19:30:18.002 I: (#0) TX 868.00Mhz | 15 05 80
19:30:18.053 I: (#0) request timeout: 2ms
19:30:18.053 I: (#0) TX 868.00Mhz | 56 15 02
19:30:18.055 -----
19:30:18.056 I: (#0) Radio infos: t: 97, s: 0, f: 0, n: 97 | p: 20
19:30:18.056 I: (#0) TX 868.00Mhz | 15 0b 80
19:30:18.057 W: CMT TX failed, code: 2
19:30:18.118 I: (#0) request timeout: 2ms
19:30:18.118 I: (#0) TX 868.00Mhz | 56 15 02
19:30:18.120 -----
19:30:18.120 I: com loop duration: 121ms
19:30:18.121 -----
19:30:23.000 I: (#0) Radio infos: t: 98, s: 0, f: 0, n: 98 | p: 20
19:30:23.001 I: (#0) TX 868.00Mhz | 15 05 80
19:30:23.069 I: (#0) request timeout: 2ms
19:30:23.069 I: (#0) TX 868.00Mhz | 56 15 02
19:30:23.071 -----
19:30:23.072 I: (#0) Radio infos: t: 99, s: 0, f: 0, n: 99 | p: 20
19:30:23.072 I: (#0) TX 868.00Mhz | 15 0b 80
19:30:23.073 W: CMT TX failed, code: 2
19:30:23.104 I: (#0) request timeout: 2ms
19:30:23.104 I: (#0) TX 868.00Mhz | 56 15 02
19:30:23.106 -----
19:30:23.106 I: com loop duration: 107ms
19:30:23.107 -----

BenjiKimba, 2024-08-24:
I read it here somewhere before,  but when using a CMT2300A module, you need a longer antenna, not a Wifi antenna  (2,4Ghz), the same one as used on a LoRa module( 1Ghz ). 
I think it must be written in the description for this project, just as the Elco for the NRF24L01+ module.
Under some circumstances, it is very critical to align a WiFi antenna on a CMT2300A module.

Alf-Ele, 2024-08-24:
Thanks for that hint. I'm using a dedicated 868MHz Antenna. I'm back to the 0.8.83, that workes fine. 0.8.102 stoppt on WLAN after a day or two....

BenjiKimba, 2024-08-24:
I am using dev. 0.8.127 with CMT2300A and HMS 1600-4t and its working fine, on different builds and software versions.

TheSCC1992, 2024-09-06:
habe ebenfalls das Problem. Was ist nu der entscheidende Unterschied wieso es ab >8.83 nicht mehr möglich ist die HMS abzufragen? In den Changelogs habe ich nicht diesbezüglich gefunden. Wäre schön wenn jemand das beantworten könnte.

Alf-Ele, 2024-09-06:
Doch in den change logs sind zwei Einträge zum CMT2300A gelistet, daher auch mein Betreff mit CMT.
Es mögen auch Verbesserungen sein, die aber bei schwachem Empfang dazu führen, dass nichts mehr empfangen wird.....

TheSCC1992, 2024-09-06:
ok, es ist aber immer noch ein Bug CMT Inverter nicht erreichbar [Bug] #1553 vorhanden welcher bisher nicht gefixt wurde.

TheSCC1992, 2024-09-15:
bekommt man hier noch eine weiterführende Antwort warum man den HMS Serie nicht mehr abfragen kann?

Alf-Ele, 2024-09-16:
Schau einmal #1749, noch jemand wo die 140er mit dem CMT nicht läuft. 
Ich bin zurück auf 83 und warte bis jemand vom CMT bug überzeugt ist