lumapu/ahoy #1734: Frage zu unterschiedlichen FW Varianten (*.bin, *all.bin und *minimal.bin)

CK1-CK, 2024-08-21:
Hallo alle zusammen

ich habe eine grundsätzliche Frage zu den unterschiedlichen FW Varianten.

Englische
- esp8266.bin
- esp8266-all.bin
- esp8266-minimal.bin 
- esp8266-prometheus.bin

Deutsche
- esp8266-de.bin
- esp8266-all-de.bin
- esp8266-prometheus-de.bin

Wie unterscheiden sich die Unterschiedlichen FW Varianten untereinander?

Speziell würden mich die **Unterschiede** zwischen der *.bin, *all.bin und *minimal.bin interessieren.

Leider habe ich dazu keine Doku gefunden.

Viele Dank schon im Voraus.
Grüße
CK1

knickohr, 2024-08-21:
Aus dem Changelog ist zu entnehmen :

https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md#0892---2024-03-10

Gubi2023, 2024-08-21:
und wo ist der Unterschied zwischen .bin und minimal.bin? gibts da echt keine Doku zu?

Gubi2023, 2024-08-21:
siehe auch #1735

CK1-CK, 2024-08-22:
Vielen Dank für die Links.

Ich habe es versucht nochmal übersichtlich dazustellen.

|       FW Tag       | MqTT  | Display | History/Verlauf | Prometheus |
|--------------------|-------|---------|-----------------|------------|
| esp8266            | X     | -       | -               | -          |
| esp8266-all        | X     | X       | X               | -          |
| esp8266-minimal    | -     | -       | -               | -          |
| esp8266-prometheus | ?     | ?       | ?               | X          |

knickohr, 2024-08-22:
Diese Frage kann wahrscheinlich nur @lumapu beantworten.

lumapu, 2024-08-22:
Danke für die Tabelle, ich habe sie vervollständigt:

> |       FW Tag/Feature       | MqTT  | Display | History/Verlauf | Prometheus |
> |--------------------|-------|---------|-----------------|------------|
> | esp8266            | X     | -       | -               | -          |
> | esp8266-all        | X     | X       | X               | -          |
> | esp8266-minimal    | -     | -       | -               | -          |
> | esp8266-prometheus | X     | X       | X.              | X          |

Diese Datei ist die beste Beschreibung der eingebauten Features:

https://github.com/lumapu/ahoy/blob/main/src/platformio.ini

CK1-CK, 2024-08-23:
Ich denke es ist alles gesagt.

Hier ist die Lösung:
https://github.com/lumapu/ahoy/issues/1734#issuecomment-2305244571

Eventuell kann man diese Informationen in die Doku mit aufnehmen.

Grüße
CK1

closed 2024-08-23 by CK1-CK