lumapu/ahoy #1736: ESP32 mit Wlan langsamer als mit W5500

Edibo63, 2024-08-21:
## Hardware
  * [ ] ESP8266
  * [ X] ESP32
  * [ ] Raspberry Pi
### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ X] external antenna (SMA)

### Version / Git SHA: 
Version: 0.1.141
Github Hash: _______

### Build & Flash Method:
* [ X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

Habe 2 baugleiche ESP32 mit nRF24L01+. Einer mit Wlan und einer mit W550.
Mit ist aufgefallen dass ab 0.8.140 der ESP32 mit Wlan einiges länger braucht als mit früheren Versionen um eine Seite aufzubauen oder von Daten zu Verlauf wechseln oder die Loginseite anzuzeigen.
Der mit W5500 ist genauso schnell wie mit älteren Versionen. Am Wlan kanns nicht liegen, da hat sich nichts  geändert.
Ist das sonst noch jemanden aufgefallen oder liegt das Problem irgendwo bei mir?

rmayergfx, 2024-08-22:
W550 oder meinst du W5500 Ethernet Shield?

Edibo63, 2024-08-24:
Tippfehler, natürlich den W5500 ethernet.
Der ESP über Wlan reagiert verzögert, wie wenn er anderweitig beschäftigt ist. War vor Vers. 0.8.140 nicht so.

lumapu, 2024-08-25:
Ich kann das nicht bestätigen. Was sagt denn unter `/system` unter der Überschrift `Verbindung` die WiFi RSSI? Bei mir steht sie bei -71dBm.

Edibo63, 2024-08-26:
-77dBm  Verbindung über einen Repeater -Abstand 2 Stockwerke: normale Reaktion
-53 dBm Verbindung direkt mit der Fritzbox- geringer Abstand : verzögerte Reaktion
-54dBm Verbindung über den Repeater- geringer Abstand zum Repeater: verzögerte Reaktion 
Das Problem scheint bei mit zu sein. Muß noch ein paar Tests machen.

lumapu, 2024-08-28:
ok, bin gespannt was rauskommt. Evtl. kannst du auch einen Ping auf die IP des ESP machen, damit hast du dann eine Latenzmessung.

Interessant, habe einen ping von 105ms gemessen. RSSI ist heute auf -69dBm, insgesamt würde ich auch behaupten, dass es sich nicht so flüssig wie eine Ethernetvariante anfühlt.

```
64 bytes from 192.168.1.41: icmp_seq=1 ttl=64 time=84.7 ms
64 bytes from 192.168.1.41: icmp_seq=2 ttl=64 time=74.7 ms
64 bytes from 192.168.1.41: icmp_seq=3 ttl=64 time=105 ms
64 bytes from 192.168.1.41: icmp_seq=4 ttl=64 time=112 ms
64 bytes from 192.168.1.41: icmp_seq=5 ttl=64 time=29.7 ms
64 bytes from 192.168.1.41: icmp_seq=6 ttl=64 time=161 ms
64 bytes from 192.168.1.41: icmp_seq=7 ttl=64 time=112 ms
64 bytes from 192.168.1.41: icmp_seq=8 ttl=64 time=109 ms
64 bytes from 192.168.1.41: icmp_seq=9 ttl=64 time=224 ms
64 bytes from 192.168.1.41: icmp_seq=10 ttl=64 time=65.5 ms
64 bytes from 192.168.1.41: icmp_seq=11 ttl=64 time=123 ms
64 bytes from 192.168.1.41: icmp_seq=12 ttl=64 time=105 ms
64 bytes from 192.168.1.41: icmp_seq=13 ttl=64 time=130 ms
64 bytes from 192.168.1.41: icmp_seq=14 ttl=64 time=56.8 ms
64 bytes from 192.168.1.41: icmp_seq=15 ttl=64 time=88.0 ms
^C
--- 192.168.1.41 ping statistics ---
15 packets transmitted, 15 received, 0% packet loss, time 35ms
rtt min/avg/max/mdev = 29.743/105.357/223.630/44.296 ms
```

zum Vergleich per Ethernet auf einen ESP32-S3 POE, 12ms im Schnitt:

```
ping -n 15 10.20.1.200

Ping wird ausgeführt für 10.20.1.200 mit 32 Bytes Daten:
Antwort von 10.20.1.200: Bytes=32 Zeit=7ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=10ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=18ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=9ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=14ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=9ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=28ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64
Antwort von 10.20.1.200: Bytes=32 Zeit=12ms TTL=64

Ping-Statistik für 10.20.1.200:
    Pakete: Gesendet = 15, Empfangen = 15, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 7ms, Maximum = 28ms, Mittelwert = 12ms
```

Edibo63, 2024-08-29:
Ethernet

PING 192.168.1.36 (192.168.1.36) 56(84) Bytes an Daten.
64 Bytes von 192.168.1.36: icmp_seq=1 ttl=64 Zeit=1.37 ms
64 Bytes von 192.168.1.36: icmp_seq=2 ttl=64 Zeit=1.05 ms
64 Bytes von 192.168.1.36: icmp_seq=3 ttl=64 Zeit=1.21 ms
64 Bytes von 192.168.1.36: icmp_seq=4 ttl=64 Zeit=1.28 ms
64 Bytes von 192.168.1.36: icmp_seq=5 ttl=64 Zeit=1.63 ms
64 Bytes von 192.168.1.36: icmp_seq=6 ttl=64 Zeit=1.56 ms
64 Bytes von 192.168.1.36: icmp_seq=7 ttl=64 Zeit=1.52 ms
64 Bytes von 192.168.1.36: icmp_seq=8 ttl=64 Zeit=1.54 ms

^C
--- 192.168.1.36 ping-Statistik ---
16 Pakete übertragen, 16 empfangen, 0% packet loss, time 15026ms
rtt min/avg/max/mdev = 1.039/1.430/1.931/0.210 ms

wlan  ESP32#1   -53dBm

PING 192.168.1.5 (192.168.1.5) 56(84) Bytes an Daten.
64 Bytes von 192.168.1.5: icmp_seq=1 ttl=64 Zeit=1011 ms
64 Bytes von 192.168.1.5: icmp_seq=3 ttl=64 Zeit=455 ms
64 Bytes von 192.168.1.5: icmp_seq=2 ttl=64 Zeit=1480 ms
64 Bytes von 192.168.1.5: icmp_seq=7 ttl=64 Zeit=1268 ms
64 Bytes von 192.168.1.5: icmp_seq=8 ttl=64 Zeit=244 ms
64 Bytes von 192.168.1.5: icmp_seq=9 ttl=64 Zeit=58.9 ms
64 Bytes von 192.168.1.5: icmp_seq=10 ttl=64 Zeit=395 ms
64 Bytes von 192.168.1.5: icmp_seq=11 ttl=64 Zeit=104 ms
64 Bytes von 192.168.1.5: icmp_seq=12 ttl=64 Zeit=127 ms
64 Bytes von 192.168.1.5: icmp_seq=13 ttl=64 Zeit=148 ms
^C
--- 192.168.1.5 ping-Statistik ---
13 Pakete übertragen, 10 empfangen, 23.0769% packet loss, time 12133ms
rtt min/avg/max/mdev = 58.898/529.068/1480.071/499.640 ms, pipe 2

wlan ESP32#2  -53dBm

PING 192.168.1.38 (192.168.1.38) 56(84) Bytes an Daten.
64 Bytes von 192.168.1.38: icmp_seq=1 ttl=64 Zeit=240 ms
64 Bytes von 192.168.1.38: icmp_seq=3 ttl=64 Zeit=205 ms
64 Bytes von 192.168.1.38: icmp_seq=2 ttl=64 Zeit=1233 ms
64 Bytes von 192.168.1.38: icmp_seq=4 ttl=64 Zeit=205 ms
64 Bytes von 192.168.1.38: icmp_seq=5 ttl=64 Zeit=1110 ms
64 Bytes von 192.168.1.38: icmp_seq=6 ttl=64 Zeit=197 ms
64 Bytes von 192.168.1.38: icmp_seq=9 ttl=64 Zeit=2261 ms
64 Bytes von 192.168.1.38: icmp_seq=10 ttl=64 Zeit=1237 ms
64 Bytes von 192.168.1.38: icmp_seq=11 ttl=64 Zeit=213 ms
^C
--- 192.168.1.38 ping-Statistik ---
13 Pakete übertragen, 9 empfangen, 30.7692% packet loss, time 12140ms
rtt min/avg/max/mdev = 197.212/766.881/2261.317/693.601 ms, pipe 3

wlan    ESP32#2   -66dBm

PING 192.168.1.38 (192.168.1.38) 56(84) Bytes an Daten.
64 Bytes von 192.168.1.38: icmp_seq=2 ttl=64 Zeit=926 ms
64 Bytes von 192.168.1.38: icmp_seq=7 ttl=64 Zeit=937 ms
64 Bytes von 192.168.1.38: icmp_seq=6 ttl=64 Zeit=1961 ms
64 Bytes von 192.168.1.38: icmp_seq=9 ttl=64 Zeit=816 ms
64 Bytes von 192.168.1.38: icmp_seq=8 ttl=64 Zeit=1832 ms
64 Bytes von 192.168.1.38: icmp_seq=11 ttl=64 Zeit=1802 ms
64 Bytes von 192.168.1.38: icmp_seq=12 ttl=64 Zeit=778 ms
64 Bytes von 192.168.1.38: icmp_seq=10 ttl=64 Zeit=2818 ms
^C
--- 192.168.1.38 ping-Statistik ---
14 Pakete übertragen, 8 empfangen, 42.8571% packet loss, time 13169ms
rtt min/avg/max/mdev = 778.407/1483.811/2818.316/687.911 ms, pipe 3

In der Fritzbox wird bei wlan eine Geschwindgkeit von 120Mbit/s  angezeigt.
Manchmal läuft es gefühlt normalschnell. Die meiste Zeit geht alles verzögert.

Edibo63, 2024-08-30:
Nach mehreren Versuchen -Abbruch wegen Zeitüberschreitung-  ist ein Update auf 0.1.143 gelungen
Jetzt läuft es wieder gut.

PING 192.168.1.5 (192.168.1.5) 56(84) Bytes an Daten.
64 Bytes von 192.168.1.5: icmp_seq=1 ttl=64 Zeit=82.3 ms
64 Bytes von 192.168.1.5: icmp_seq=2 ttl=64 Zeit=101 ms
64 Bytes von 192.168.1.5: icmp_seq=3 ttl=64 Zeit=122 ms
64 Bytes von 192.168.1.5: icmp_seq=4 ttl=64 Zeit=149 ms
64 Bytes von 192.168.1.5: icmp_seq=5 ttl=64 Zeit=172 ms
64 Bytes von 192.168.1.5: icmp_seq=6 ttl=64 Zeit=194 ms
64 Bytes von 192.168.1.5: icmp_seq=7 ttl=64 Zeit=11.0 ms
64 Bytes von 192.168.1.5: icmp_seq=8 ttl=64 Zeit=33.6 ms
64 Bytes von 192.168.1.5: icmp_seq=9 ttl=64 Zeit=56.6 ms
64 Bytes von 192.168.1.5: icmp_seq=10 ttl=64 Zeit=76.5 ms
^C
--- 192.168.1.5 ping-Statistik ---
10 Pakete übertragen, 10 empfangen, 0% packet loss, time 9013ms
rtt min/avg/max/mdev = 11.045/99.902/194.488/56.663 ms

Edibo63, 2024-09-11:
Nachdem mit 0.8.143 alles wieder mit normaler Geschwindigkeit kann man hier zumachen
Danke

closed 2024-09-11 by Edibo63

stefan123t, 2024-09-11:
@lumapu was hatte sich denn zwischen 0.8.140 und 0.8.143 geändert dass es jetzt Faktor fünf bis zehn schneller ist bei @Edibo63 ?
Auffällig auch dass sein Ethernet Faktor zehn schneller ist als Deins (oder meines) ;)

lumapu, 2024-10-07:
müsste im Changelog stehen.