lumapu/ahoy #1737: 0.8.140 / Reboot issue / Task Watchdog

Alpha4Linux, 2024-08-23:
### Platform

ESP32

### Assembly

I did the assembly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.140

### Github Hash

f1f4481

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT is on
1 x Inverter HM800

### Error description

0.8.83 was running fine. after upgrade to 0.8.140 i can see reboots on a daily basis.
Reboot reason: Task Watchdog
Ahoy is connected to local WLAN

JWRu, 2024-08-24:
Same for me:
Platform ESP32 (bought from eBay shop "nordicskills")
Version 0.8.140
OTA update from 0.8.83
MQTT is on
1 inverter HM-800

From time to time the DTU freezes completely (website not accessible) and I have to power cycle it.

lumapu, 2024-08-25:
are you able to share a Coredump? Go to /system, on the botton of the page you'll find a button `download CoreDump`

Alpha4Linux, 2024-08-25:
/coredump is empty here
{}
Reboot reason	Task Watchdog

JWRu, 2024-08-25:
>   /coredump is empty here
>   {}
>   Reboot reason Task Watchdog

exactly the same for me.

fr-500, 2024-08-25:
Like reported on Discord same for me

roku133, 2024-08-25:
Similar experience with ESP8266 - Software/System restart about twice a day.
v0.8.83 has worked more than 60 days without any interruption. 👍
Configuration:
OTA update from v0.8.83 to ESP8266 standard version v0.8.140
MQTT enabled
1 inverter HM-400

lumapu, 2024-08-28:
so viele Berichte über das wohl scheinbar gleiche Problem. Was ich nicht verstehe, warum beim ESP32 der Coredump leer ist. Stürzt der EPS währendessen ab?
Könnt ihr bitte noch bisschen was zur Konfiguration sagen? MqTT, WiFi Empfang gut? Wechselrichter weit entfernt?

JWRu, 2024-08-28:
**MqTT**
connected: true
#TX: 6566
#RX: 0

**WiFi** RSSI: -54 dBm

**Inverter** 5 m Luftlinie - RSSI: -65 dBm

Zickerl, 2024-08-28:
Servus, bei mir läuft es momentan stabil. Beziehe jetzt die Zeit über die Fritzbox per IP.Allerdings ist mir aufgefallen, dass die Zeit im Sekundenintervall blinkt. Hab aber 8266. Mqtt nicht benutzt. WR in der Nähe - wifi okGesendet mit der GMX Mail AppAm 28.08.24 um 21:14 schrieb Lukas Pusch
                
            
            
                
                    Von: "Lukas Pusch" ***@***.***>Datum: 28. August 2024An: "lumapu/ahoy" ***@***.***>Cc: "Zickerl" ***@***.***>,"Comment" ***@***.***>Betreff: Re: [lumapu/ahoy] 0.8.140 / Reboot issue / Task Watchdog (Issue #1737)
                
                
so viele Berichte über das wohl scheinbar gleiche Problem. Was ich nicht verstehe, warum beim ESP32 der Coredump leer ist. Stürzt der EPS währendessen ab?
Könnt ihr bitte noch bisschen was zur Konfiguration sagen? MqTT, WiFi Empfang gut? Wechselrichter weit entfernt?

Alpha4Linux, 2024-08-28:
Mein Setup
Inverter: HM-800
ESP32-D0WDQ6-V3: Distanz zum Inverter 20cm
MQTT: On (Broker im selben IP-Range, Port 1883, Topic: hm800)
WLAN aus Sicht AP: Signal Strength: -56 dBm, Speed: 72 Mbps, Channel Width: 20 MHz

roku133, 2024-08-28:
Configuration:
Inverter: HM-400
ESP8266 - Wemos D1 mini
Distance DTU - Inverter: 5 m
MQTT: on (#TX 163443 / #RX 0)
WLAN: RSSI: -72, CH: 11, 65 Mbit/s

[image]

lumapu, 2024-08-28:
wow, so schnell so viele Antworten 👍. Wie steht das Intervall zur Inverterabrage? Insgesamt sieht ein TX-Count von 6000+ nicht schlecht aus.
Auch beim ESP8266 sieht die Heap-Fragmentation von 3 sehr gut aus.

Läuft bei einem eine Nulleinspeisung, oder in anderen Worten eine zyklische Abfrage der API?

JWRu, 2024-08-28:
>  Wie steht das Intervall zur Inverterabrage?
60 Sekunden

> Läuft bei einem eine Nulleinspeisung, oder in anderen Worten eine zyklische Abfrage der API?
Nein - ich nutze nur MqTT

roku133, 2024-08-28:
Strange: after several Software/System restarts as described above currently uptime more than three days! 
Every 30 seconds the values are updated, no cyclic queries, only MqTT.

Zickerl, 2024-08-29:
Guten Morgen Lukas,

vorab, ich bin technisch nicht der Beste und kenne mich mit den Dingen eher weniger aus.

Aber ich liefere dir hier nochmal andere Daten.

Inverter - HM600
ESP8266 auf Wemos D1 Mini V 3.0.0 - 4 MB
WIFI RSSI: -58
Datenempfang RSSI: -64 dBm
Distanz zum Inverter ca. 5 m
Intervall der Abfrage unverändert bei 15 Sekunden
Keine Nulleinspeisung

Wie berichtet ist es in der Darstellung (Handy oder PC) so, dass unter AHOYDTU die ESP-Anzeige im Sekundentakt klickt.
Alle 10 Sekunden zucken dann auch die Sätze "Wechseltrichter ... und Sie verwenden ..." - vermutlich erfolgt a eine Abfrage.
Obwohl als Takt 15 Sekunden hinterlegt sind.
Unter Einstellungen - NTP-Server habe ich ja IP-Adresse meiner FB eingetragen und seitdem funktioniert es wieder.
Aber auch hier blinkt die Systemzeit. Port und Intervall sind auf 123 und 720 unverändert.

Verwende das Teil nur um zu kontrollieren ob der WR arbeitet, da das Teil bereits schonmal getauscht wurde.

Schönen Gruß und danke für dein Engagement
Alfred

revo-net, 2024-09-04:
Hi,
bei mir hängt sich der Controller nicht auf, nur das Netzwerk ist dann weg.
Ich habe einen 8266 via USB an einem raspi und habe die serielle migeloggt.
Wenn das Netzwerk ausfällt, läuft die serielle normal weiter, zeigt aber auch keine zusätzlichen Informationen. Der Inverter wird weiterhin abgefragt...
Aber nach einer längeren Zeit gibt es dann eine OOM Exception und der Controller bootet neu und alles beginnt vor vorne.
Nach dem Deaktivieren von mqtt scheint der Fehler nicht mehr auszutreten...
BG,
rene

007-mk, 2024-09-08:
bei mir gibt es ebenfalls Neustarts. meine Systemkonfig:
ESP8622 + Display + Funkmodul + Kondensator
- Wemos D1 Mini
- NRF24L01+
- Kondensator 100uF 
- Display SH1106 1.3"

Version 0.8.140 / 0.8.143 Fehler tritt bei beiden Versionen auf (jeweils die all-Version verwendet). Habe die lite und normale Version ebenfalls getestet, scheinen stabiler zu laufen.
OTA update from 0.8.83
MQTT ist an
1 inverter HM-800
ein Python-Script ruft alle 15 Sekunden rest API auf.
Aufrufen der Weboberfläche und Klicken auf einzelnen menüpunkte triggert meistens ein Neustart

dknubben, 2024-09-12:
Ich habe ebenfalls seit einigen (Dev-)Versionen spontane Reboots festgestellt (ESP32, Task Watchdog, leider ohne Coredump). Moment läuft 0.8.143. Nicht tragisch, aber merkwürdig und zuerst nicht nachvollziehbar. Die Intervalle der Reboots finden im Bereich von mehreren Stunden bis 1-2 Tagen statt. Nach Abschalten von MQTT ist das Problem eindeutig gelöst und die DTU läuft nun seit einer Woche ohne Probleme (8 Inverter, NRF24L01+ und 100uF, kein Display).
MQTT wird zwar nicht genutzt und die Daten wurden lediglich an einen Mosquitto-Server gesendet, der immer verfügbar war. Alle sonstigen Abfragen erfolgen über die REST-API.

007-mk, 2024-09-12:
> Ich habe ebenfalls seit einigen (Dev-)Versionen spontane Reboots festgestellt (ESP32, Task Watchdog, leider ohne Coredump). Moment läuft 0.8.143. Nicht tragisch, aber merkwürdig und zuerst nicht [...]

Seit dem ich heute morgen MQTT deaktiviert habe, hatte ich keinen Reboot mehr. Scheint am MQTT zu liegen.

Alpha4Linux, 2024-09-15:
Hab nun mal zum Test MQTT ausgeschalten (Broker IP-Adresse gelöscht)
Ahoy-DTU hat nun Uptime: 2 days, 20:39:15.
Scheint also ein Issue mit MQTT zu sein.

007-mk, 2024-10-02:
> Seit dem ich heute morgen MQTT deaktiviert habe, hatte ich keinen Reboot mehr. Scheint am MQTT zu liegen.

Meine Beobachtung nach einiger Zeit: Es kommt immer noch zu reboots aber deutlich weniger oft. Sobald ich die Seite im Browser aktualisiere, kommt es zum Reboot.

Alpha4Linux, 2024-10-02:
Ahoy-DTU 0.8.140 seit der Deaktivierung von MQTT stabil; Uptime 19 days, 14:47:12
Kommunikation nur über REST-API

dknubben, 2024-10-02:
> > Seit dem ich heute morgen MQTT deaktiviert habe, hatte ich keinen Reboot mehr. Scheint am MQTT zu liegen.
> 
> Meine Beobachtung nach einiger Zeit: Es kommt immer noch zu reboots aber deutlich weniger oft. Sobald ich die Seite im Browser aktualisiere, kommt es zum Reboot.

Das wird nach meiner Meinung eine andere Ursache haben. Ich würde an Deiner Stelle auf den ESP32 umstellen. Probleme mit dem ESP8622 sind hier in den Issues schon mehrfach thematisiert worden.

lumapu, 2024-10-03:
die `0.8.150` hat eine kleine Änderung, ich weiß allerdings nicht, ob sie wirkt. Es kann durch die Änderung nur besser werden.

dknubben, 2024-10-03:
> die `0.8.150` hat eine kleine Änderung, ich weiß allerdings nicht, ob sie wirkt. Es kann durch die Änderung nur besser werden.

Vielen Dank! Ich habe nun mit der .151 MQTT reaktiviert und werde das mal beobachten und ggf. Rückmeldung geben.

dknubben, 2024-10-04:
Leider keine Verbesserung, eher im Gegenteil. Innerhalb der letzten ca. 14 Std. mindestens 2 Reboots mit aktiviertem MQTT.

lumapu, 2024-10-04:
@dknubben hast du einen Coredump für mich?

Gubi2023, 2024-10-04:
auch bei mir: nach 12 Std reboot der .151 mit task watchdog
De2 2 Std.-Versatz ist auch noch vorhanden, siehe screenshot
[image]

@lumapu:  coredump im discord

dknubben, 2024-10-04:
> @dknubben hast du einen Coredump für mich?

Leider nein, im Webinterface ist der Coredump wie immer leer.
Über serielle Verbindung wird das geloggt:

`abort() was called at PC 0x4011812c on core 0
Backtrace: 0x40083ca9:0x3ffbecec |<-CORRUPTED
ELF file SHA256: 0a5ef494daf3d085
E (11608) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
`

Weiss nicht, ob das hilfreich ist. "flash config" irritiert mich. Ohne MQTT ist ja alles in Ordnung.

JWRu, 2024-10-06:
Auch bei mit gibt's mit .151 Reboots - Coredump ist leer: {}

lumapu, 2024-10-07:
echt komisch, dass die coredumps leer sind. Habt ihr schon versucht über den Webflasher euer System neu aufzusetzen?

JWRu, 2024-10-07:
@lumapu
> Habt ihr schon versucht über den Webflasher euer System neu aufzusetzen?

Habe ich jetzt mal gemacht. Der Webflasher installiert Version .83 - habe dann ein OTA-Update auf .151 gemacht.
Werde jetzt weiter beobachten.

juepi, 2024-10-09:
Hallo Leute,

Zur Info: der AhoyDTU reboot mit reason "Task Watchdog" passiert immer noch, wenn MQTT aktiviert ist und der Broker nicht erreichbar ist. Ich verwende zur Zeit die `0.8.151`.

lg,
Jürgen

dknubben, 2024-10-09:
> Hallo Leute,
> 
> Zur Info: der AhoyDTU reboot mit reason "Task Watchdog" passiert immer noch, wenn MQTT aktiviert ist und der Broker nicht erreichbar ist (d.h. während des Betriebs ausfällt, weil z.b. der Server nach einem Update rebootet).
> 

Das ist leider nicht korrekt. Der Reboot passiert auch dann sporadisch, wenn der Broker verfügbar ist.

Gubi2023, 2024-10-09:
bei mir gibt es auch reboots, obwohl MqTT deaktiviert ist 😒

juepi, 2024-10-09:
> Zur Info: der AhoyDTU reboot mit reason "Task Watchdog" passiert immer noch, wenn MQTT aktiviert ist und der Broker nicht erreichbar ist (d.h. während des Betriebs ausfällt, weil z.b. der Server nach einem Update rebootet). Ich verwende zur Zeit die `0.8.151`.

Etwas schlecht formuliert, also nochmal: der einzige reboot von AhoyDTU der **bei mir** seit der Installation der Version `0.8.151` (am 3.10.) auftrat passierte nachdem der MQTT broker heruntergefahren wurde (NTP server und WiFi blieben online). Als reboot reason meldet die UI "Task Watchdog".

lg,
Jürgen

Gubi2023, 2024-10-10:
dass aktiviertes MqTT zu reboots führt, ist seit längerem bekannt...

dknubben, 2024-10-10:
> dass aktiviertes MqTT zu reboots führt, ist seit längerem bekannt... Nur merkwürdig, dass es keinen Coredump gibt. Vllcht PC-Problem? (Bin-Dateien speichern blockiert/gesperrt?)

Definitiv nein, siehe auch die gleichen Meldung von anderen. Welches PC-Problem meinst Du konkret? Der Coredump wird als JSON im Browser (bei mir Firefox) geöffnet, falls vorhanden.
Da es mit der .152 MqTT -Änderungen gab, habe ich die Rückmeldung gegeben.

Gubi2023, 2024-10-11:
komisch, bei mir haben die coredumps format .bin... (024-10-10_10-54-32_v0.8.152_esp32-wroom32-de_coredump.bin)

dknubben, 2024-10-11:
> komisch, bei mir haben die coredumps format .bin... (024-10-10_10-54-32_v0.8.152_esp32-wroom32-de_coredump.bin) und werden in meinem Download-Ordner gespeichert.

Ich weiss nicht, ob wir vom gleichen Coredump reden. Im Webinterface über System - download CoreDump wird mir die (beim mir leere bzw. nur mit Klammern {} versehene) Datei coredump.json ausgegeben. Wie kommst Du an die .bin-Datei?

Gubi2023, 2024-10-11:
Bin-Datei wird in meinem Download-Ordner gespeichert.

dknubben, 2024-10-11:
> Bin-Datei wird in meinem Download-Ordner gespeichert. siehe Video
> Aufzeichnung.2024-10-12.002233.mp4
> 
> so sieht es mit firefox aus. Mit chrome wird die Datei nur nach Aufforderung gespeichert. Welchen browser nutzt du?

Danke für die schnelle Rückmeldung. Ich nutze auch den Firefox. dieser öffnet dann http://IP_AHOYDTU/coredump als JSON.  Das ist auch so bei Chrome. Eine Webconsole (in Firefox mit CTRL-SHIFT-I) gibt auch für die Datei "Content-Type | application/json; charset=utf-8" aus. Was zeigt Dir die .bin-Datei in einem Texteditor an?

Gubi2023, 2024-10-12:
[2024-10-12_00-48-27_v0.8.152_esp32-wroom32-de_coredump.txt](https://github.com/user-attachments/files/17349049/2024-10-12_00-48-27_v0.8.152_esp32-wroom32-de_coredump.txt)

Gubi2023, 2024-10-12:
http://IP_AHOYDTU/coredump sagt bei mir "Seite nicht gefunden"

JWRu, 2024-10-12:
Vorhin habe ich endlich mal einen Reboot "mitschneiden" können:
Ich habe die DTU an meinen Laptop gehängt, Serial Debug eingeschaltet und CoolTerm mitlaufen lassen.
Hier ist der Ausschnitt aus dem Log:
[Log_Reboot.txt](https://github.com/user-attachments/files/17349268/Log_Reboot.txt)

JSpannhacke, 2024-10-15:
Hab das selbe Problem. GIT SHA: f1f4481 :: 0.8.140 ESP32, rebooter sich teilweise mehrfach am Tag und hat sich nun schon mehrfach auch einfach komplett aufgehäng, dann hilft nur Stecker ziehen. Leider kein Core dump. MQTT ist an und ich gehe ab und zu auf die Weboberfläche. Erstaunlich oft ist er genau dann gerade neu gestarten.

dknubben, 2024-10-15:
> http://IP_AHOYDTU/coredump sagt bei mir "Seite nicht gefunden"
> 
> Webconsole kann ich nicht interpretieren

IP_AHOYDTU ist selbsverständlich nur ein Platzhalter für die individuelle IP Deiner Ahoy DTU. Die Webconsole kannst mit aktuellen Browsern über die genannte Tastenkombination aufrufen und Dir dann den Netzwerkverkehr und die Header der heruntergeladenen Dateien ansehen. Ich wollte lediglich wissen, woher Deine .bin-Datei stammt.

haraldhotzbehofsits, 2024-10-18:
Seit 152 installiert ist, seit 8 Tagen kein reboot

Gubi2023, 2024-10-18:
> Seit 152 installiert ist, seit 8 Tagen kein reboot

Hast du MqTT am laufen?

revo-net, 2024-10-21:
Bei mir läufts nun einier Zeit schon seit der .150
Frage derzeit nur einen Inverter ab.
Finde die mqtt Infos sollte man tweaken können, braucht man ja selten im debug style ;)
---------------------------------
mqtt: #TX | 1370404
17 days, 19:11:11
Software/System restart
esp8266 (build options: M-en)
0.8.150 - 677876a
CPU: 80MHz, undefined Core(s)
---------------------------------

JSpannhacke, 2024-10-22:
.152 läuft bei mir mit 6 Wechselrichter und MQTT nun seit 2.5 Tagen ohne Reboot

Parsival944, 2024-10-23:
Der zyklische Reset stoppte bei meinem Modul wenn der MQTT Abfrage Intervall von 0 (Default) auf 600 geändert wurde.

Gubi2023, 2024-10-23:
bin ich ja gespannt, obs bei mir auch was nützt. Bisher fast täglichen Reboot gegen 8Uhr....

dknubben, 2024-10-23:
> Der zyklische Reset stoppte bei meinem Modul wenn der MQTT Abfrage Intervall von 0 (Default) auf 600 geändert wurde.

Danke für den Hinweis. Das wäre aber für mich trotzdem keine dauerhafte Lösung, weil dann die Intervalle zur Datenerfassung viel zu gross sind.
Generell sprechen wir aber alle von mehreren unterschiedliche Reboot-Ursachen. Die "Task Watchdog"-Reboots treten bei mir nach Deaktivierung von MQTT nicht mehr auf. Dafür gibt es immer wieder mal "Panic"-Reboots (aktuell Version .152). Dem bin ich noch nicht nachgegangen. Die "8 Uhr"-Reboots von @Gubi2023 konnte ich allerdings so noch nie feststellen.

JWRu, 2024-10-24:
Habe auch seit gestern .152 installiert. Heute Nacht gab's einen Reboot - gleiches Bild wie vorher:
[Log_Reboot_02.txt](https://github.com/user-attachments/files/17502216/Log_Reboot_02.txt)

Gubi2023, 2024-10-24:
> Der zyklische Reset stoppte bei meinem Modul wenn der MQTT Abfrage Intervall von 0 (Default) auf 600 geändert wurde.

na ja, hat immerhin einen Tag länger durchgehalten, Reboot nach ca 50 statt 24 Std. 🤔

CLAM01, 2024-10-25:
> > Der zyklische Reset stoppte bei meinem Modul wenn der MQTT Abfrage Intervall von 0 (Default) auf 600 geändert wurde.
> 
> Danke für den Hinweis. Das wäre aber für mich trotzdem keine dauerhafte Lösung, weil dann die Intervalle zur Datenerfassung viel zu gross sind. Generell sprechen wir aber alle von mehreren unterschiedliche Reboot-Ursachen. Die "Task Watchdog"-Reboots treten bei mir nach Deaktivierung von MQTT nicht mehr auf. Dafür gibt es immer wieder mal "Panic"t bei mqtt-Reboots (aktuell Version .152). Dem bin ich noch nicht nachgegangen. Die "8 Uhr"-Reboots von @Gubi2023 konnte ich allerdings so noch nie feststellen.

Mit meinem esp8266 könnte ich die von gubi dargestellten reboots auch nachstellen. Pünktlich zum Start der Produktion und dadurch erhöhten last bei mqtt ist der Ahoy Matrose zusammen gebrochen um nach einer Panic oder exeption 29 oder OOM neu zu starten.

JSpannhacke, 2024-10-26:
Beim mir gab es auch wieder trotz .152 ein Reboot. Für mich sieht es so aus als würde irgendwo ein pointer fremden Speicher überschreiben.

Gubi2023, 2024-10-27:
hatte soeben einen Reboot meines ESP32 zum Zeitpunkt der Meldung "Time calibration" des HM600. Ob das miteinander zu tun hat? @lumapu:  Coredump im discord

belzig, 2025-01-14:
Hab gestern mqtt disabled und seitdem kein Einfrieren mehr.
Leider war es dadurch nötig mein Nulleinspeistungs Skript auf rest-api umzuschreiben.

Wenn jemand sowas brauchen sollte: >
https://forum.domoticz.com/viewtopic.php?p=322775#p322775

[7 more replies]