lumapu/ahoy #1743: bug: update von .140 auf .142 schlägt fehl

Gubi2023, 2024-08-28:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an image of my wiring

### Version

0.8.142

### Github Hash

565a69e

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update auf .142

### Error description

update von .140 auf .142 funktioniert nicht. DTU bekommt keine Verbindung zum NTP-Server und startet sich alle 15 sec. neu.
Neustartgrund panic,  coredump anbei

Gubi2023, 2024-08-28:
[image]

Gubi2023, 2024-08-29:
Dauerreboot! DTU konnte nur über USB webflasher mit erase wieder zum laufen gebracht werden

reserve85, 2024-08-29:
kann ich bestätigen, .142 ist bei mir nicht funktionsfähig.

lumapu, 2024-08-29:
sorry, das war unbeabsichtigt. Danke für eure reports und Geduld mit mir 😇

Gubi2023, 2024-08-29:
0.8.143: funzt! danke für den schnellen Bugfix 😊👍

DanielR92, 2024-09-01:
close?

Gubi2023, 2024-09-01:
ich hab ihn mal offen gelassen, falls andere auch über diesen bug stolpern...l

lumapu, 2024-09-02:
ich denke nicht, dass allzuviele die 142 installiert haben, daher mache ich zu.

closed 2024-09-02 by lumapu